CrealityOfficial/CrealityPrint #4: Wen support blocker feature?

Lancelight, 2023-09-30:
This slicer is missing a crucial element to "auto supports" called a Support blocker.  Without this,  the auto support feature is literally completely unusable.

xZILFx, 2023-10-06:
this is needed

Lancelight, 2023-12-12:
Looks like this slicer and the K1/K1 Max as a result is DOA (Dead on arrival).  It's based on a Cura thats so old that no one has any interest in modifying it because all they would be doing is nothing but wheel re-invention.  Creality really F'd up on the K1/K1 Max rollout with this garbage thats for sure.  It's to bad too,  the hardware itself is great.  But without a decent slicer,  it's literally just an unusable paperweight.  I'm so glad I got rid of my K1 Max now.  I had doubts that his project would actually get anywhere and here we are,  end of the year and not a SINGLE line of actual code has been added to this repo.  The issues queue is just piling up with absolutely no hope of any of them actually being completed.

closed 2025-04-26 by github-actions[bot]