CrealityOfficial/CrealityPrint #5: Slicer Timings

jinxy1234, 2023-09-30:
The print time is a little out, the time estimate on the attached was 38 + hours, it took 27 + some minutes. The layer height (story) also shows as 0, the other models were sliced with an earlier version.

The picture was a screen grab just before the print finished.
[image]

leseaw, 2023-10-06:
same here, older version works better.

hemiao2023, 2023-10-17:
try to upgrade firmware.

tPayne0647, 2024-05-07:
Its very buggy. Sometimes slice estimation times are way longer and printer prints super slow. I have to close and reopen. I sliced the same file a few times and got wildy different times. i kept getting 4.5 hr estimation then 11hr, closed the program and resliced and then got 1 hr.

closed 2025-04-26 by github-actions[bot]