CrealityOfficial/CrealityPrint #9: new filament saving problem

AllenTM5, 2023-10-14:
I noticed a problem with saving own filament profiles and also with updating generic filaments, after creating a new filament from generic copy there is no option to change filament manufacturer, after changing filament settings and pressing save, after restarting creality print the filament in override settings has all options selected and after each startup and before printing with a given filament you have to manually change these options because they do not want to be saved.

hemiao2023, 2023-12-05:
The problem of filament profiles has been optimized in version 4.3.8，and we will completely solve the problem of parameter configuration in version 4.4.

AllenTM5, 2023-12-05:
> The problem of filament profiles has been optimized in version 4.3.8，and we will completely solve the problem of parameter configuration in version 4.4.

ok so where i can download version 4.3.8 and when it will be published 4.4 version

closed 2025-04-26 by github-actions[bot]