CrealityOfficial/CrealityPrint #13: Creality Print laggy on Mac M1

botolo, 2023-10-23:
Creality Print is very laggy after slicing a model on Macbook Air M1. Other slicers (Cura, Prusa Slicer and Bambu Studio) are very smooth after slicing a model: you can rotate the sliced model, check different layers, etc. Creality Print makes this very very difficult. For example, if I zoom a sliced model, it takes 3 to 5 seconds for the model to zoom. I have seen the same issue on Simplify3D.

hemiao2023, 2023-12-05:
We are currently analyzing the Macbook Air M1 adaptation issue.

protopopescu, 2023-12-07:
In addition to seeing this on my M1 Mac Mini, I see this problem also on my 2014 Intel Core i5 Mac Mini, so it might not be only an M1 issue. The graphical interface of Creality Print is very laggy, while Prusa Slicer has no issues, as described above.

spitfire, 2024-02-08:
> We are currently analyzing the Macbook Air M1 adaptation issue.

It is only compiled for intel, so it might be a (rosetta) emulation issue. Can you try to compile it as a universal binary?

botolo, 2024-02-09:
I’m sorry but unfortunately I have no idea how to compile source code into a universal app.

spitfire, 2024-02-10:
> I’m sorry but unfortunately I have no idea how to compile source code into a universal app.

Sorry, I didn't mean you, but @CrealityTech who owns this app.

protopopescu, 2024-02-12:
The lag issue that I have reported for v4.3.7 seems to have been solved in v4.3.8. I am running v4.3.8.6984 on both my M1 and Intel Macs, without experiencing any of the previously reported lag.

botolo, 2024-02-12:
I am using Creality Print v4.3.8.6984 on a MacBook Air M1 with 16GB RAM and I still have major lagging issues. This is Creality Print.

https://github.com/CrealityOfficial/CrealityPrint/assets/3123816/0fadc0bf-62df-4914-bdb5-3a8b5d1128ab

This is Prusa Slicer with the same model...100% smooth.

https://github.com/CrealityOfficial/CrealityPrint/assets/3123816/bf6b324d-6cb5-4d1b-a3b7-430930948206

spitfire, 2024-02-12:
> I am using Creality Print v4.3.8.6984 on a MacBook Air M1 with 16GB RAM and I still have major lagging issues. This is Creality Print.
> 
>  Creality.Print.mov 
> 
> This is Prusa Slicer with the same model...100% smooth.
> 
>  Prusa.Slicer.mov

same here. And if I put models in it it becomes incredibly sluggish

closed 2025-04-26 by github-actions[bot]