CrealityOfficial/CrealityPrint #14: AppImage cannot slice model on Fedora

hansegard, 2023-10-29:
With the latest AppImage, Creality_Print-v4.3.7.6631-x86_64-Release.AppImage, slicing is not possible on Fedora 39. When pressing Slice, nothing happens and this errors shows up in the console log

`SensorsAnalytics SDK send failed: Timeout was reached`

These are the records flushed

`flush record : [{"_track_id":29285270,"distinct_id":"1C:1B:B5:E1:95:B0","event":"BuyTicket","lib":{"$lib":"cpp","$lib_method":"code","$lib_version":"1.0.2"},"properties":{"$lib":"cpp","$lib_version":"1.0.2","app_name":"Creality Print","platform":"Win"},"time":1698434805270,"type":"track"},{"_track_id":757738756,"distinct_id":"1C:1B:B5:E1:95:B0","event":"BuyTicket","lib":{"$lib":"cpp","$lib_method":"code","$lib_version":"1.0.2"},"properties":{"$lib":"cpp","$lib_version":"1.0.2","app_name":"Creality Print","platform":"Win"},"time":1698434838756,"type":"track"},{"_track_id":437685661,"distinct_id":"1C:1B:B5:E1:95:B0","event":"BuyTicket","lib":{"$lib":"cpp","$lib_method":"code","$lib_version":"1.0.2"},"properties":{"$lib":"cpp","$lib_version":"1.0.2","app_name":"Creality Print","platform":"Win"},"time":1698434865661,"type":"track"},{"_track_id":896212632,"distinct_id":"1C:1B:B5:E1:95:B0","event":"BuyTicket","lib":{"$lib":"cpp","$lib_method":"code","$lib_version":"1.0.2"},"properties":{"$lib":"cpp","$lib_version":"1.0.2","app_name":"Creality Print","platform":"Win"},"time":1698434872632,"type":"track"},{"_track_id":455307857,"distinct_id":"1C:1B:B5:E1:95:B0","event":"BuyTicket","lib":{"$lib":"cpp","$lib_method":"code","$lib_version":"1.0.2"},"properties":{"$lib":"cpp","$lib_version":"1.0.2","app_name":"Creality Print","platform":"Win"},"time":1698434877857,"type":"track"},{"_track_id":6743213,"distinct_id":"1C:1B:B5:E1:95:B0","event":"BuyTicket","lib":{"$lib":"cpp","$lib_method":"code","$lib_version":"1.0.2"},"properties":{"$lib":"cpp","$lib_version":"1.0.2","app_name":"Creality Print","platform":"Win"},"time":1698435533213,"type":"track"},{"_track_id":290385310,"distinct_id":"1C:1B:B5:E1:95:B0","event":"BuyTicket","lib":{"$lib":"cpp","$lib_method":"code","$lib_version":"1.0.2"},"properties":{"$lib":"cpp","$lib_version":"1.0.2","app_name":"Creality Print","platform":"Win"},"time":1698435545310,"type":"track"},{"_track_id":221159214,"distinct_id":"1C:1B:B5:E1:95:B0","event":"BuyTicket","lib":{"$lib":"cpp","$lib_method":"code","$lib_version":"1.0.2"},"properties":{"$lib":"cpp","$lib_version":"1.0.2","app_name":"Creality Print","platform":"Win"},"time":1698435549214,"type":"track"},{"_track_id":697303349,"distinct_id":"1C:1B:B5:E1:95:B0","event":"BuyTicket","lib":{"$lib":"cpp","$lib_method":"code","$lib_version":"1.0.2"},"properties":{"$lib":"cpp","$lib_version":"1.0.2","app_name":"Creality Print","platform":"Win"},"time":1698435593349,"type":"track"},{"_track_id":945115248,"distinct_id":"1C:1B:B5:E1:95:B0","event":"BuyTicket","lib":{"$lib":"cpp","$lib_method":"code","$lib_version":"1.0.2"},"properties":{"$lib":"cpp","$lib_version":"1.0.2","app_name":"Creality Print","platform":"Win"},"time":1698437885248,"type":"track"},{"_track_id":107266828,"distinct_id":"1C:1B:B5:E1:95:B0","event":"BuyTicket","lib":{"$lib":"cpp","$lib_method":"code","$lib_version":"1.0.2"},"properties":{"$lib":"cpp","$lib_version":"1.0.2","app_name":"Creality Print","platform":"Win"},"time":1698437906828,"type":"track"},{"_track_id":657385915,"distinct_id":"1C:1B:B5:E1:95:B0","event":"BuyTicket","lib":{"$lib":"cpp","$lib_method":"code","$lib_version":"1.0.2"},"properties":{"$lib":"cpp","$lib_version":"1.0.2","app_name":"Creality Print","platform":"Win"},"time":1698437915915,"type":"track"},{"_track_id":211610335,"distinct_id":"1C:1B:B5:E1:95:B0","event":"BuyTicket","lib":{"$lib":"cpp","$lib_method":"code","$lib_version":"1.0.2"},"properties":{"$lib":"cpp","$lib_version":"1.0.2","app_name":"Creality Print","platform":"Win"},"time":1698437920335,"type":"track"},{"_track_id":638575321,"distinct_id":"1C:1B:B5:E1:95:B0","event":"BuyTicket","lib":{"$lib":"cpp","$lib_method":"code","$lib_version":"1.0.2"},"properties":{"$lib":"cpp","$lib_version":"1.0.2","app_name":"Creality Print","platform":"Win"},"time":1698437935321,"type":"track"},{"_track_id":920690096,"distinct_id":"1C:1B:B5:E1:95:B0","event":"BuyTicket","lib":{"$lib":"cpp","$lib_method":"code","$lib_version":"1.0.2"},"properties":{"$lib":"cpp","$lib_version":"1.0.2","app_name":"Creality Print","platform":"Win"},"time":1698438080096,"type":"track"},{"_track_id":827324310,"distinct_id":"1C:1B:B5:E1:95:B0","event":"BuyTicket","lib":{"$lib":"cpp","$lib_method":"code","$lib_version":"1.0.2"},"properties":{"$lib":"cpp","$lib_version":"1.0.2","app_name":"Creality Print","platform":"Win"},"time":1698438804310,"type":"track"},{"_track_id":564114729,"distinct_id":"1C:1B:B5:E1:95:B0","event":"BuyTicket","lib":{"$lib":"cpp","$lib_method":"code","$lib_version":"1.0.2"},"properties":{"$lib":"cpp","$lib_version":"1.0.2","app_name":"Creality Print","platform":"Win"},"time":1698438844729,"type":"track"},{"_track_id":484200445,"distinct_id":"1C:1B:B5:E1:95:B0","event":"BuyTicket","lib":{"$lib":"cpp","$lib_method":"code","$lib_version":"1.0.2"},"properties":{"$lib":"cpp","$lib_version":"1.0.2","app_name":"Creality Print","platform":"Win"},"time":1698438860445,"type":"track"},{"_track_id":983033862,"distinct_id":"1C:1B:B5:E1:95:B0","event":"BuyTicket","lib":{"$lib":"cpp","$lib_method":"code","$lib_version":"1.0.2"},"properties":{"$lib":"cpp","$lib_version":"1.0.2","app_name":"Creality Print","platform":"Win"},"time":1698578853862,"type":"track"}]`

hemiao2023, 2023-12-05:
Thanks for your feedback.
The 4.3.8 Linux version is expected to be released this week, and we hope it will solve your problem.

pchevali, 2023-12-12:
When will this 4.3.8 version will be released ? 
The same problem occurs with ubuntu ( And  is present also with the 3 previous releases ).
Is there at least a  workaround to run it on linux ?

pchevali, 2023-12-14:
It works for me now with 4.3.8 on ubuntu 22.04 !

Sanva, 2024-01-15:
I was having the same problem with 4.3.7 ( latest version linked in https://www.crealitycloud.com/software-firmware/software/creality-print ) in Linux Mint 21.3.

4.3.8 seems to work fine.

ronoaldo, 2024-05-25:
I can confirm 4.3.8 worked for me on Fedora as well. Website download links should be updated to point to this download.

closed 2025-04-26 by github-actions[bot]