CrealityOfficial/CrealityPrint #15: Feature request: Create CrealityPrint flatpak

hansegard, 2023-10-29:
Please create and deploy a flatpak of CrealityPrint. Flatpaks are platform agnostic and they are generally more portable and stable as compared to AppImages. The flatpak manifest could probably look very similar to that of [Cura Flatpak](https://github.com/flathub/com.ultimaker.cura).

JaneSmith, 2024-01-14:
+1. I'd like this too. My operating system is Fedora Silverblue, and Flatpak is essentially the native package manager. Currently only Cura, PrusaSlicer, and now Bambu Studio are available on Flathub - no other slicers.

watzon, 2024-10-05:
Seeing as CrealityPrint is a fork of Bambu Studio, their Flatpak would probably be a good starting point for anyone that wants to tackle this. You can find it here https://github.com/flathub/com.bambulab.BambuStudio

ClaytonRehder, 2025-04-07:
Good point, all the other major slicers already have a Flatpak version: OrcaSlicer, BambuStudio, PrusaSlicer, Cura. Most of them are available in Flathub as well.