CrealityOfficial/CrealityPrint #27: Traduction en français

NOX56390, 2023-11-28:
Bonjour 
Je souhaite avoir une traduction en français. 
J'ai essayé de voir s il était possible de le faire .par moi même. Mais je n'ai pas assez de connaissances pour changer les mots dans le programme. 
Est t il possible d avoir un accès au programme pour faire une traduction simple en français ? 
Ou plus simplement.  Y a t' il une version de prévu en français  ?

beatwinthewave, 2023-11-29:
@NOX56390  Je ne sais pas si Creality a prévu de s'en charger, mais il y a bien les mécanismes pour le faire.

Grosso-modo, dans ce dépôt github, il y a un dossier "locales" qui contient les fichiers qui permettent la traduction du logiciel. Il y a notamment un tableau contenant quelques 900 lignes pour chaque "texte" utilisé dans l'interface de l'application. Ce n'est pas impossible d'obtenir une traduction en français, et ça pourrait etre automatiser mais ça s'annonce tout de même légèrement fastidieux. Certains utilisateurs ont déjà utilisés ces fichiers pour proposer des traductions dans leurs langues à travers une "Pull request".  

Ce n'est pas chose facile d'expliquer en quelques mots "comment participer dans les règles de l'art de la programmation" à un projet comme celui-ci, mais pour commencer, il aurait été plus pratique de poser la question en anglais afin qu'elle soit comprise par un public international.

NOX56390, 2023-11-30:
@beatwinthewave 
Merci de m avoir répondu. 
OK, je découvre  github. Je doit avouer que je ne comprends pas tout ! 
Effectivement je comprends qu'une demande en anglais serait mieux . 
Je vais regarder " pull request " .
Normalement "quand on cherche on trouve "

hemiao2023, 2023-12-05:
We will add multi-language adaptation in version 4.3.9, includingFrench.，and this work is in progress.
Regarding the accuracy of some translations, we would like your help, and we will put the translation files on github.

NOX56390, 2023-12-05:
OK
 Envoyez un lien pour les fichiers.  Je vais essayer de vous aider ...
Merci

NOX56390, 2023-12-29:
@hemiao2023    @beatwinthewave
bonjour 
merci pour la traduction en français ! je travail à amélioré cette traduction .j'ai presque fini ,reste a me corriger!
j'ai une demande de tirage ( en brouillon ), est t'il possible de visualiser mes modifications (uniquement pour moi !) dans le soft , car je pense qu'il va fa falloir adapter en fonction de la traduction .

merci