CrealityOfficial/CrealityPrint #31: Control panel in Printer Information detail view

protopopescu, 2023-12-07:
The 'Control' panel in the Printer Information detail view shows controls that are irrelevant/not available for my printer. For instance, in the case of CR-10 SE, and I suppose most other 'open' printers (i.e. not K1*) the 'Side fan', 'Back fan' and 'Led switch' are not operational.

In such case, i.e. when the printer viewed does not support the above features, one could move the 'Model fan' dialog up on the same line with the position info/dialogs, and hide the 'Side fan', 'Back fan' and 'Led switch', thus leaving more room for the 'File' list below.

hemiao2023, 2023-12-16:
Thank you for your good suggestion. We will discuss it

protopopescu, 2024-03-22:
Thank you, I see that some of the suggestions have been implemented: the 'Side Fan' and 'Back Fan' do not show up for my CR-10 SE which is not equipped with side and back fans. 

However, the 'Led switch' is there but does nothing. On the CR-10 SE there's a physical switch for the light bar. Is the 'Led switch' ON/OFF supposed to work when the physical switch is ON?

closed 2025-04-26 by github-actions[bot]