CrealityOfficial/CrealityPrint #43: Error trying to Compile On Raspberry Pi 4

mrlinux2u, 2023-12-18:
Running `python3 ./cmake/ci/cmake.py -c -b -e --channel_name=opensource` produces the following error message on my Raspberry Pi 4 (running Bullseye 32bit).

> _parse_args error. 

Is there any software I need to install apart from Python3 before I can compile under Linux (FYI running the pre-compiled Linux version is not an option as I only have Raspberry Pi 4/5's)?

mrlinux2u, 2023-12-20:
This exact same issue occurs when running a Ryzen 5 desktop with the 64bit version of Mint Linux (work PC) - it would be good if a list of all required libraries etc. was included in the compile instructions because at the moment it doesn't appear to be possible to compile CrealityPrint (has anyone tried on Windows?) from source.

hemiao2023, 2023-12-21:
We are fixing some compilation scripts

mrlinux2u, 2023-12-21:
Is there any possibility for you to also provide a pre-compiled ARM64 version (there are over 40 million Raspberry Pi's and a lot of users use them for 3D printing)?

mfraser, 2023-12-27:
Has anyone been able to compile CrealityPrint for any variant of Linux using the instructions given?

savek-cc, 2023-12-30:
I have been trying to compile it using fedora core 39 for a couple of hours.
After doing a recursive checkout, installing conan 1.62 and running the ci command, it actually starts building.
However, the included sources generate different compiler errors:

- trimesh2: missing a <cstdint.h> include in at least 2 places
- tbb_static: gets a warning on "changes meaning" for "task" - and as it's apparently using -Werror, it is aborting compilation.

gcc version 13.2.1

ernani, 2024-08-12:
Trying branch main and 5.0.3 branch but no success. Problem seems to be related to scripts! Any ETA to get this fixed? Or then please, provide us with an AppImage release of CP 5.0

closed 2025-04-26 by github-actions[bot]