CrealityOfficial/CrealityPrint #44: Creality Print 4.3.8 keeps crashing.

AndyShap2020, 2023-12-20:
I've used 4.3.7 with no issues since I installed it. You fixed an issue with 4.3.7 in 4.3.8, so I installed right away. It keeps crashing (specifically locking up) every time I use it. This can happen as  soon as 5 minutes after I open it.  It is completely unresponsive. The only way to fix it is to end task.

hemiao2023, 2023-12-21:
Where did you download the installation package from? the github version we upgrade the QT version to 5.15.2, I'm not sure if it will cause any problems.
Official website installation package use the QT 5.13, you can try it.
https://file2-cdn.creality.com/file/f0606a25bbbd76e6032f4c349a0f61d3/Creality_Print-v4.3.8.6984-win64-Release.exe

hemiao2023, 2023-12-21:
> Right from this page. I'll try the link you gave me. I just tried and it says I do not have access to that page. :-(

https://www.creality.com/pages/download-software?spm=..index.header_1.1

AndyShap2020, 2023-12-21:
It looks like I have version 6986 form the GitHub page. I'll try 6984 from the link you just sent me.  Thank you.

AndyShap2020, 2023-12-22:
It looks like moving to 6984 fixed the issue! Thank you!

Chandler-Kluser, 2024-12-21:
> > Right from this page. I'll try the link you gave me. I just tried and it says I do not have access to that page. :-(
> 
> https://www.creality.com/pages/download-software?spm=..index.header_1.1

this version is crashing under linux, too. I have been using PrusaSlicer with a community made configuration file for almost the whole year because Creality doesn't give a good support for customers.

I would complain this version you mentioned is crashing, but it is just f*cking slow it made me believe it crashes. It takes many minutes to open a Save Project... dialog, this is unacceptable, there is no terminal messages to help me try to identify what kind of issue is that

closed 2025-04-26 by github-actions[bot]