CrealityOfficial/CrealityPrint #45: Initial Layer Horizontal Expansion

frankt2012, 2023-12-20:
I think I found a bug in Creality Print 4.3.8. It seems initial layer horizontal expansion is not working at all. No matter if I set it to a negative or positive value it does nothing. I need to have it at -0.35 to compensate for elephants foot. If I reinstall the old version it works perfectly.

Sebi3D, 2023-12-22:
I am having this problem as well with regular horizontal expansion. Others have reported flow having no affect specifically Ironing Flow. Though I have not look into that specific one.

frankt2012, 2023-12-22:
> I am having this problem as well with regular horizontal expansion. Others have reported flow having no affect specifically Ironing Flow. Though I have not look into that specific one.

I sent them an email to the address in the about section of Creality Print and they said there will be an update on github.

closed 2025-04-26 by github-actions[bot]