CrealityOfficial/CrealityPrint #46: Big Slowdown  on 4.3.8

AllenTM5, 2023-12-23:
I noticed a big slowdown in version 4.3.8, a simple model like the one in the screenshot below was slicing about 30 minutes into the time I took the screenshot

[image]

Cblaze401, 2023-12-26:
Same here, painfully slow.

hemiao2023, 2023-12-27:
You can send a simple test stl and gcode file to us. May be related to some parameters. Our email crealityprint@creality.com

Cybotage, 2024-04-08:
Same here. Sliced a model in Orca in under a minute, More or less same settings in Creality and takes forever and a day. Orca is using multiple cores. Seems Creality is not. All my cores are snoozing as Creality is "busy" slicing

closed 2025-04-26 by github-actions[bot]