CrealityOfficial/CrealityPrint #53: Initial Print Temperature 4.3.9 Beta

frankt2012, 2024-01-01:
Initial Print Temperature 4.3.9 beta is not working. Not sure if it was working in previous versions. I wanted to raise it up to get a better first layer but it prints at 220 no matter what. I set the Initial Print Temperature to 230.

kashank, 2024-03-06:
@frankt2012 I just ran into this as well on 4.3.7.6627. You're actually looking for "Printing Temperature Initial Layer", which apparently can only be found when toggling "advanced" mode and looking for it. **However** setting that value still does nothing as it doesn't actually persist when the "save" button is clicked:
[image]

kashank, 2024-03-06:
Found a workaround - exporting a printer profile and then adding the line:

`material_print_temperature_layer_0=*your value here*`

Then re-importing the profile will allow you to see and edit "Printing Temperature Initial Layer" for that profile.  This might be an issue with how the slicer treats the built-in profiles.

frankt2012, 2024-03-06:
> Found a workaround - exporting a printer profile and then adding the line:
>
> `material_print_temperature_layer_0=*your value here*`
>
> Then re-importing the profile will allow you to see and edit [...]

Thank you. I have to check that out. Hopefully they will fix this in a new beta for 4.3.9. The first beta they posted works great for me except this issue and bridging seem to be broken. I hope they post a new one soon. I see there where updates after the last beta but can't figure out how to compile it in Windows.

closed 2025-04-26 by github-actions[bot]