CrealityOfficial/CrealityPrint #59: Ender 3 V3 SE/KE Nozzle Sizes

zbergus, 2024-01-19:
Ender 3 V3 SE/KE missing nozzle sizes other than 0.4, I have 0.2 and 0.6 nozzles from Creality but have to modify a bunch of stuff manually just to use them due to lack of ability to chose a nozzle size

Rikajlo, 2024-02-14:
The CR-10 SE also has only a 0.4mm nozzle. There should be at least an option to save my profile for different diameters and export it to share with the world if officially supporting different sizes is impossible.
:^   )

AnachronicNomad, 2024-12-30:
Bumping this.  I print almost exclusively with 0.6mm and 0.8mm for my business, was very excited to use the auto-level on the V3 KE.  Come to find out, none of the open-source slicers have a profile made for the printer yet, let alone specifics for the nozzle size.  The auto-level on the V2 Neo I have keeps giving out, I just wasted 300g of filament on a mis-print.  

Is there help-documentation we can edit to more easily keep a profile around for different nozzle sizes in the Creality Print slicer?

closed 2025-04-26 by github-actions[bot]