CrealityOfficial/CrealityPrint #64: Compile Error "assert trimesh2 exist."

CalDymos, 2024-01-28:
I Get an Error if i Try to compile the source code on windows with 

 **.\cmake\ci\cmake.py -c -b -e --channel_name=opensource** 

I used Windows 10 x64 and
Git-2.43.0-x64
cmake-3.28.1-x64
python-3.9.0-x64
conan-1.55-x64
VS 2019 (v16.6.2)

Error Message:
...
...
-- CONAN ****** Start Find cxlog
-- __conan_import cxlog [cxlog]
-- CONAN ****** End Find cxlog
-- __enable_spdlog failed.
-- CONAN ****** Start Find trimesh2
-- __conan_import trimesh2 [trimesh2]
-- CONAN ****** End Find trimesh2
CMake Error at cmake/ConfigureTarget.cmake:781 (message):
  assert trimesh2 exist.
Call Stack (most recent call first):
  msbase/CMakeLists.txt:5 (__assert_target)

-- Configuring incomplete, errors occurred!
ERROR     RUN FAILED.
********************************************
...
...

ShixinSun, 2024-01-29:
Compiling does not work for me either!

Error message:
...
...
CMake Error at cmake/ConfigureTarget.cmake:781 (message):
  assert trimesh2 exist.
Call Stack (most recent call first):
  msbase/CMakeLists.txt:5 (__assert_target)

-- Configuring incomplete, errors occurred!

System:
Win 10 x64
Conan 1.55
python 3.12.1
Visual Studio 16.11.33
cmake 3.28.1

CalDymos, 2024-01-30:
Has anyone been able to get the project compiled under Windows, and if so, how?

macfaulkner, 2024-04-11:
Won't compile for me either on macOS Sonoma 14.4.1 / M1. Sounds like it's not an OS issue. 

I'm going through some of the config options, I'll post if I'm able to find anything.

ssouthurst, 2024-09-27:
I've spent quite a while trying to figure this out - not knowing Python, and having never used Cmake (WTF - use MSVC...).

I'm getting very close now with the following...

I'm using VS2107 (because I can't be bothered updating on the laptop I'm currently on) - so I had to call:

```batchfile
SET VS_VERSION=Visual Studio 15 2017
```
This might not be an issue for others.

I found that trying to compile would spew a bunch of errors, then close so I used:

```batchfile
python -i <command>
``` 

Which stops it at the end with a python console - lets you scroll back through the errors.  I found that it wasn't able to create a log file, which I fixed by changing the logger in cmake.py:

```python
import logging
logger = logging.getLogger(__name__)
logging.basicConfig(filename='example.log', encoding='utf-8', level=logging.DEBUG)
```

This helped heaps - was getting a lot of "The current directory is not valid" which I traced to the calls to git to fetch the various libs (including trimesh).

After A LOT of stuffing around I found that this was due to the "Microsoft Store" installation of Python - which runs python in a sandbox.  So when the code in ci_conan.py is creating folders, the folders aren't created in the "Roaming\circle_conan_github_cache" path - but they are created in "Local\Packages\Python...\circle_conan_github_cache" folder - which Git can't find.

So I uninstalled python, installed from the msi at Python.org - and suddenly (after resetting the VS_VERSION env var) it started - almost - working...

I'm now up to "'conan' is not recognized as an internal or external command,"

Which is odd - because I have installed it.  Probably a search path issue.  

The struggle goes on...

Phypo, 2024-10-03:
Hello, 
I  tryed to compile CrealityPrint on linux mint, there are many problems because cmake files is too old
- In openmp.cmake   
-> line 19 , I remplace it with: 	set(OPENMP_ROOT /usr/lib/llvm-14)
and line 22, I remplace  it with if(NOT EXISTS ${OPENMP_ROOT}/include/ompt-multiplex.h)

- cmake are not found the trimesh2 lib, I found it on Github and compile it but I don't know how to tell the initialization script where it is.

for ssouthurst , conan is a python lib you must install it with :
 pip install conan

 for [ssouthurst] for conan it's a python libra

ssouthurst, 2024-10-07:
Yep - it's installed - it just can't find it...  Path etc. is all OK - it's being a pain.

I'm currently seeing if I can get it working as a Visual Studio project.

closed 2025-04-26 by github-actions[bot]