CrealityOfficial/CrealityPrint #74: Can not PAN the view on a MAC

Dochartaigh, 2024-04-05:
On a Windows computer, it seems you can hold shift and click and move the image up/down/left/right (or right click and drag somebody said).

On a MAC, this seems impossible (and it's not just me, plenty posts asking how to do this...). This makes Creality Print slicer absolutely useless because I can't see what I need to see in the 3d render window...

Are all us Mac users missing something silly? PLEASE let us know what it is!

Up to date Creality Print software, on Mac Mini  M2 Pro, running Sonoma OS.

raycardillo, 2024-05-27:
I'm using version 5.0.3.8736 on WINDOWS and I cannot pan the view using SHIFT and Click of the mouse either (first or second button). This is really annoying and makes it very difficult to investigate the preview for example.

SargFowler, 2024-07-12:
Seriously Creality? You can't pan on a Mac with the trackpad?
Sort out the flickering, slow slicing and jerky zoom while you're on.
Otherwise people are just going to use other slicers.

bozhich, 2024-09-29:
I'm not sure if this is still relevant but on MacOS You could  move the image up/down/left/right while holding spacebar
and rotate the camera with control (or right click if you have it enabled)

carlosjunod, 2025-03-28:
> I'm not sure if this is still relevant but on MacOS You could move the image up/down/left/right while holding spacebar and rotate the camera with control (or right click if you have it enabled)

I have installed Creality Print V6.0.5.1840 and 15.4 Beta MacOS and space bar is not working, any other suggestions ?