CrealityOfficial/CrealityPrint #77: Cannot compile 4.3.9 on Ubuntu - ./cmake is empty

ringstrand, 2024-04-26:
All, sorry if this is the wrong place to moan, I'm a noob on this stuff.

I downloaded the .zip-file and extracted it, but as the ./cmake folder is empty, the proposed 'pyton3 ./cmake/ci...' command of course fails.

(The 4.3.8 appimage hangs my machine after slicing (the preview tab seems to check mouse pointer once every three seconds, so it becomes very jumpy) so I wanted to try 3.9)

Running Ubuntu 22.04 on a Lenovo.

ronzle, 2024-04-27:
Hey 

clone the main project ... 
Then click in github on the cmake folder, which takes you to the 4.3.9 cmake repo. 
Clone this also

ringstrand, 2024-04-27:
Ah, thanx. (Maybe remove the .zip file as it seems a bit shitty. :)

ronzle, 2024-04-27:
The zip file is fine :) 
cmake is a dedicated repo ... so this will neither in the creality pull nor in the zipfile included 

If you are lucky to compile it, please let me know, i wasn't and gave up

ringstrand, 2024-04-27:
Aha! Chezz…
 I am too old for this shit…

I’ll try.

(But when I get error messages in line with ”Error 7, reason 43, try to
sprunk your stryffle-inhibator and open the DJE prior to clipping you TSD
or GWE” I tend to just get the whiskey bottle and switch on the tv.

Br’s

Hans.

closed 2025-04-26 by github-actions[bot]