CrealityOfficial/CrealityPrint #79: MacOS version for CP 5.0

pganns, 2024-04-29:
Hey

When will the CP5 version for macos/arm be released?

Thanks

spitfire, 2024-04-29:
Waiting for this too. As I looked at the releases I've noticed 4.3.8 got an additional arm build added later to it.

ernani, 2024-05-03:
Same thing for me!

tomaszpikus, 2024-05-05:
Hey :) I'm joining the question. Is 5.0 even planned for MacOS? If so, is there any timeline for its implementation? I'm working on MacOS and using Creality Print, which is an easy-to-use program, and I'm getting very good prints from my Ender 3 V3 KE 3D printer.

spitfire, 2024-05-05:
> Hey :) I'm joining the question. Is 5.0 even planned for MacOS? If so, is there any timeline for its implementation? I'm working on MacOS and using Creality Print, which is an easy-to-use program, and I'm getting very good prints from my Ender 3 V3 KE 3D printer.

It is pretty terrible, you should try Orca Slicer or Cura if you've rooted your printer, they have a lot better performance, provide a lot of more flexibility and are not really harder to use.

revunix, 2024-09-07:
Push!

Where is the problem for the devs to compile this as universal app??? This is unusable on an Apple Silicon Mac.

spitfire, 2024-09-07:
@CrealityTech  any update?

closed 2025-04-26 by github-actions[bot]