CrealityOfficial/CrealityPrint #84: Linux Release 5.0.0 missing

hagbardZ, 2024-04-30:
please add the appimage for linux.

esistgut, 2024-05-02:
Is there any reason to delay the Linux release?

feiticeir0, 2024-05-02:
> Is there any reason to delay the Linux release? Are you dropping the support for Linux? Please provide some insight as this kind of infos are useful when purchasing 3d printers.

I don't think they will drop Linux support. I've opened a ticket before (1 month ago) to know why the Linux version was 1 release behind the Windows and MacOS . They told me that during May, they would release new Linux version.  

I was surprised when I found out that here on Github there was a newer Linux Version - not available on the downloads page. 
So, why don't they release all the versions at the same time ? I'm guessing Windows has the bigger user base, so, it's more of a priority to them.  But they should try to release the versions all at the same time.

ernani, 2024-05-03:
Looking forward for the Linux v5 AppImage release as I use Chromium OS.

ronoaldo, 2024-05-25:
@CrealityTech - Maybe merge this with #116 and #102? It would make sense to have a single issue to track release of Linux build.

As a workaround I'm currently running the Windows build with Wine on Debian. For other places where wine is not available (such as ChromeOS/ChromiumOS) a native build would be the best option.

Also, specially to honor the Open Source license of the stack you're based on top of, release the 5.x source code as this would allow for Linux users with experience to be able to build their own.

As I understand it, the current Linux options are:
1. Build from source from 4.x branch (Cura based slicer).
2. Use the 4.8 released .AppImage (Cura based slicer).
3. Use the 5.0.3 released .exe with Wine (OrcaSlicer based slicer).
4. Use OrcaSlicer directly as it has support for most recent printers from Creality.

eduncan911, 2024-08-29:
5.x is now a clone from Orca.  and they are not going to help with anything with 4.x.

Feel free to scream in the "there's no linux release of 5.x" issue here: https://github.com/CrealityOfficial/CrealityPrint/issues/183

closed 2025-04-26 by github-actions[bot]