CrealityOfficial/CrealityPrint #85: Z-hop not turning of

joaquinlapiere, 2024-04-30:
I have turned off Z-hop in my material settings but my machine still Z-hops.
[image]
[image]
^these settings still keep it Z-hopping. if I turn the overrides off ("Z hop when retracting" "Z hop type" and all in the "Z-lift enforcement" category) the printer keeps Z hopping. 
yes the correct material is selected in the slicer. 
yes I have checked other settings. 
When using the same settings in Orca slicer (where this is based on I think), the printer doesn't Z-hop.
So I do think it is a bug and not a fault on my part. if it is, then I am sorry but after searching for 4 hours in settings I don't know what my fault could be.

I hope this helps.

joaquinlapiere, 2024-05-01:
glad to hear I'm not the only one experiencing thing, I was going a bit mad trying to find a solution. hopefully it will be fixed soon

DMaster-HU, 2024-07-27:
Almost 3 months later, in version 5.1.1 this issue still exists.

joaquinlapiere, 2024-07-29:
My printer has been out of order for a month (waiting for parts). but before that I stopped using Crealityprint and went back to orca slicer bc of issues like this. Also bc I installed Guppyscreen so I couldn't print directly from cp. Since this version of cp is clearly based on orca I don't think it will be that big of an adjustment. 

I didn't want to switch bc I like the UI of cp more than orca but orca is more reliable and has better bugfix support etc.

I hope it will be fixed someday and that I could go back to this, but I doubt that it will be soon.

closed 2025-04-26 by github-actions[bot]