CrealityOfficial/CrealityPrint #89: The mouse cursor positioning on windows is misaligned

Adilrafiq001, 2024-05-01:
[image]

The mouse cursor position on windows is misaligned. To click a button the cursor position is not aligned, I need to keep the cursor a bit above  then the original position to click a button. Also the minimized mode shows a line around the loaded app, tht is the difference of the cursor position.

dsherburn, 2024-05-06:
I have the same issue. Hope they find a fix soon
[image]

Adilrafiq001, 2024-05-07:
You need to add it to your GPU i.e. graphics card and run it on GPU and that solved my issue.

dsherburn, 2024-05-07:
How do you that exactly

Sent from my Verizon, Samsung Galaxy smartphone

closed 2025-04-26 by github-actions[bot]