CrealityOfficial/CrealityPrint #92: Program does not display correctly on Dell Precision M4800

sgraber, 2024-05-02:
When the program starts, it does not maximize properly nor does the mouse properly select anything on screen.  The mouse, when clicked, selects items about 1 cm below where the mouse is currently positioned.  Any idea why this is happening?  System information with the following resolution settings below.  As you can see from the screenshots, the mouse is well above the item its selecting, circled in red.  In the final screenshot, you can see that the program never maximizes properly either.

[image]

Device name	Dell-M4800
Processor	Intel(R) Core(TM) i7-4810MQ CPU @ 2.80GHz   2.80 GHz
Installed RAM	16.0 GB
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display
Edition	Windows 10 Pro
Version	22H2
Installed on	‎4/‎1/‎2021
OS build	19045.4355
Experience	Windows Feature Experience Pack 1000.19056.1000.0

[image]

[image]

[image]

Adilrafiq001, 2024-05-03:
I had raised a similar issue earlier. This needs an immediate fix.

Cblaze401, 2024-05-09:
Same issue as well

marekpb, 2024-05-09:
I have the same problem with Creality Print 5.0. What is wrong ? How to solve the problem?

LiscaBR, 2024-05-15:
Same problem here, Acer I7 16gb Nvidia GeForce 940M

carlos-chierici, 2024-05-30:
Same problem here with Creality Print 5.0.3.8736 running on 1920 x 1080 resolution.

af01dk, 2024-06-03:
Same problem here - very frustrating. Windows 10 on Thinkpad X1.

closed 2025-04-26 by github-actions[bot]