CrealityOfficial/CrealityPrint #96: Very slow print time on the new Ender 3 V3 Core XZ Vs. Ver.4.3.8

av-atlas, 2024-05-02:
Same model, when sliced on the new V.5 Vs. the official V4.3.8.6984, using the same Hyper PLA profile on both versions, without changing any parameter, layer height 0.2. Only compared the infill type and infill percentage to be the same on both.
The new V5 is always extremely slow. here is one example of many:

* I did perform "push parameter package update" and Updated the Ender 3 V3 to the latest version.
* I've tried it on different layers and infill. different models. the V5 is always much slower than 4.3.8
[image]

av-atlas, 2024-05-03:
I've found the error in the profile that is causing the slow printing time. Notice the extremely low values used in the new V.5 under Manage Printer>Move Ability > Acceleration limit.
After 
[image]

After I've manually updated the values to these (not all exist in V4.3.8, so some might not be correct), the printing time was reduced by 2.5 hours!
[image]

av-atlas, 2024-05-03:
I've found what is the source of the bug.
After installing V.5 the printing time is low and the values are correct, as seen below
[image]
but once I check the box for "Push Parameter package update"
[image]
and I update it:
[image]
for the new V1.2.5
[image]
the acceleration values are dramatically reduced:
[image]

Mustangmvp, 2024-05-08:
`but once I check the box for "Push Parameter package update"`

Hello @av-atlas , where did you find this option because I can't find it in my creality 5 ?

Thanks for solving the bug

av-atlas, 2024-05-09:
> `but once I check the box for "Push Parameter package update"`
> 
> Hello @av-atlas , where did you find this option because I can't find it in my creality 5 ?
> 
> Thanks for solving the bug
hope they see it and fix it :)
follow the instructions in this video from the 4:28 time mark 
https://youtu.be/159BisWUQF8?t=268

DadAlwaysWins99, 2024-11-11:
Hey guys, new to the game but late to this post... I just found another culprit to slow speeds on Reality Print V5, you need to change your Volumetric Speed. Default is 100 and it limits any and all speeds. I put it from 100 to 2000, now my V5 is like a race car with rocket fuel.

closed 2025-04-26 by github-actions[bot]