CrealityOfficial/CrealityPrint #102: Request support for Linux

artyom-intellias, 2024-05-05:
4.3.8 was provided with Appimage format, is Linux abandoned?

mikeq3, 2024-05-05:
We need a Linux release. I don't use Windows.

artyom-intellias, 2024-05-06:
For some reason i've had a hope that I will be able to build from source, but both archives contains only readme and license files, is that some kind of a joke?
@CrealityTech i'm hungry for your comments on that

[image]

hagbardZ, 2024-05-07:
Linux release is urgently needed !!!

supra89t, 2024-05-24:
Seriously, why is there no longer any linux builds? Should we sell this shit and go with bambu labs?

supra89t, 2025-04-19:
Wow... They even copied their bots! But not their Linux support.

supra89t, 2025-05-23:
https://github.com/CrealityOfficial/CrealityPrint/releases/download/v6.1.1/CrealityPrint-V6.1.1.2283-x86_64-Beta.AppImage