CrealityOfficial/CrealityPrint #107: Thumbnails not appearing on Ender 3 V3 printing from 5.0

tPayne0647, 2024-05-07:
When printing either via LAN or cloud, it seems thumbnails don't show up on my machine (Ender 3 V3) or my cloud account. Sliced files uploaded to cloud. 
[image]
[image]

Makes it a lot harder to manage files.

slibbinas, 2024-05-09:
Tumbnail are aprearing.... Problem is that Creality print 5 do not generate tumbnails at all. It always sends the defaut tumbnail. I have root and checked what it uploaded to the printer. It for every model it uploads the same png :D :D. They just did not implemented the feature :)

DMaster-HU, 2024-05-09:
> Tumbnail are aprearing.... Problem is that Creality print 5 do not generate tumbnails at all. It always sends the defaut tumbnail. I have root and checked what it uploaded to the printer. It for every model it uploads the same png :D :D. They just did not implemented the feature :)

Well, no. Actually there is the correct thumbnail in the files, only in a different format than before. That's why the printer firmware cannot parse it. Look at the gcode file with a text editor, the thumbnail blocks are different for each model (although starting the same).
You can try: copy the lines between "; thumbnail begin" and "; thumbnail end" to a text file, remove the "; " (the space too!) from the start of the lines, and feed it to a [base64-to-image](https://www.rapidtables.com/web/tools/base64-to-image.html) converter. That's why [Creality said](https://www.reddit.com/r/crealityk1/comments/1cf7ud6/incoming_firmware_update/) the new firmware will solve this problem.

tPayne0647, 2024-05-10:
You can actually just go to `manage printer` and check the advanced option. You need to select `CR_PNG` for the format and then save as a copy. Just tested and its working for me! So they just need to make that the default option I'm assuming.

nortonr3, 2024-06-14:
Hey friends! the solution is to change the size of the gcode thumbnail to 96x96,300x300 and the format of g-code thumbnail to PNG, on manage printer in advanced option.  
[image]

ViCticus, 2024-08-30:
Thanks tPayne0647, it helped me for Creality K1C
[image]

closed 2025-04-26 by github-actions[bot]