CrealityOfficial/CrealityPrint #109: Your CrealityPrint 5.0 is obvioisly reskin based on OrcaSlicer, you should not change the license to be GPL3.0 and must open source！

WeiTaoSJTU, 2024-05-08:
Please adhere to the open-source agreement and refrain from being a thief, as it greatly impacts your company's reputation.

osmolenko, 2024-05-12:
+++

hagbardZ, 2024-05-15:
still no source code :(... please provide the source code !!!

SaltWei, 2024-05-17:
[image]
Your company even claims on your official website that the software is independently developed! How shameless!

WeiTaoSJTU, 2024-05-17:
> still no source code :(... please provide the source code !!!

They have already claimed that it is self-developed.
https://www.creality.com/pages/download-software?spm=..index.header_1.1
[image]

cosine-lgray, 2024-05-24:
They do have this posted

https://github.com/CrealityOfficial/CrealityPrint

[image]

closed 2025-04-26 by github-actions[bot]