CrealityOfficial/CrealityPrint #111: CR-10 SE profile missing

Bizzik, 2024-05-10:
What the heck! Why did you add older printer (Ender 3 V3 KE) but not CR-10SE. Did you abandon CR-10 SE?

mrweisz, 2024-05-11:
I also miss the new software, if it only has 5 prints?

superior-julz, 2024-05-13:
@Bizzik 

The cr-10 SE was released simultaneously as the ender 3 v3 KE (autumn event), but they should still have both profiles.

GoodFriend1234, 2024-05-14:
Add profile for CR10SE

PHlightroom, 2024-05-14:
Please add a good profile for the CR10SE. I haven't had mine for a month yet, it's crazy the latest version of CrealityPrint doesn't support it.

Goaten22, 2024-05-14:
Not even Creality themself are able to do a good profile for the CR 10SE! I really belive so!

marcoaccasarin, 2024-05-14:
Did you forget the CR-10 SE owners? We are feeling abandoned by you

henrivdr, 2024-05-14:
@hemiao2023  Do you have a response for us please?

FEdmtecth, 2024-05-17:
i have profile for CR-M4 if someones needs , tell me.

FEdmtecth, 2024-05-17:
> Add profile for CR-10 SE

You can configure it in orcaslicer , and next you can create your own profile. I made this for CR-M4 :)

SebACDA, 2024-12-04:
> i have profile for CR-M4 if someones needs , tell me.

I would be very grateful if You could share it with me.

FEdmtecth, 2024-12-04:
In this moment , the profile from CR-M4 and the machine is added in Orcaslicer lastest versions.

SebACDA, 2024-12-04:
Thank You.
I was trying to use the Creality Slicer 5.1.6
Would You recommend Orca ?

FEdmtecth, 2024-12-04:
Don´t use Creality Slicer 5.1.6 it´s for this printer , it´s my advise .
I use Creality Print V4.3.8.7058 or OrcaSlicer 2.2.0 .
Test Both and see what is the best for you .

SebACDA, 2024-12-04:
I will.
Thank You very much !!!

FEdmtecth, 2024-12-04:
welcome

closed 2025-04-26 by github-actions[bot]