CrealityOfficial/CrealityPrint #116: 5.0.2 Linux Release ?

hagbardZ, 2024-05-14:
Please add Linux release !!!

CaiMaochang, 2024-05-14:
Please add MacOS ARM64 release, thank you so much.

ronoaldo, 2024-05-15:
+1 for the Linux build! Alternatively, kindly publish the newer version in source form so we can build ourselves.

As a workaround, I'm currently using both Creality Cloud (very very beginner friendly for my wife and kids!) and OrcaSlicer, which worked super well with K1 Max.

allangj, 2024-05-16:
+2 in the linux release please, part of the community wants to give it a try. Or at least release the code and we can help and contribute if you guys are having trouble with a linux release

feiticeir0, 2024-05-17:
Why are you, Creality, keep releasing versions, but still won't publish a Linux version of the v5 version ? Why ?

mathojojo, 2024-05-19:
+1. Creality Print 4.3.7.6627 never worked on linux, and I never got any answer about that from the creality developpers. So I have many hopes about the 5.x.x versions... But they are still missing ! Please, considere it as a priority now.

Liionboy, 2024-05-19:
They don't give a fck for what we need, you don't see guys they ignore us?

ronoaldo, 2024-05-25:
We surely need Creality to release either source code or the Linux build, but as I see, they seems to be incrementally fixing issues in the 5.x track for the currently supported platforms. I'm not defending them but at least they work is quite visible with the recent releases.

As a workaround for this issue i was able to slice and print (including network print!) with the 5.0.3 windows release using Wine, on Debian 12 KDE with WineHQ wine version 9.0.

As a side note: I do like some of the changes made by Creality to the Orca/Prusa base they used in v5. Some are not very cool! I hope that once they release, the same features can be ported to other slicers as well regarding the UI.

feiticeir0, 2024-05-25:
They keep releasing windows and mac versions (yesterday they released version 5.0.3), but Linux still keeps getting behind... Latest Linux version is 4.3.8.6991... WTF  Creality .... F you
And to make maters worst, I'm still using version 4.3.7 because 4.3.8 doesn't has the CR-PLA-Carbon fillament. The App still doesn't has the K1C printer as one of the options..  I'm using K1 profile...

ernani, 2024-08-08:
I've tried to build myself the linux version with no success:

I get the following error:

INFO      Run: conan create --profile linux -s build_type=Debug /tmp/tmpa12nrnbr opensource/linux
ERROR: conanfile.py doesn't declare package name or version

closed 2025-04-26 by github-actions[bot]