CrealityOfficial/CrealityPrint #120: Nozzle size selection is missing when adding printer on Creality Print 5

rxfxt, 2024-05-16:
When adding printer in Creality Print 5.0.2, only 0.4 mm nozzle size is available:

[image]

Nozzle sizes 0.6 mm and 0.8 mm are missing, which were previously avaiable in last Creality Print 4 (4.3.8)

LuboSta, 2024-05-17:
any update ? cant use potential of my k1 max on this slicer without it ! creating new printer and selecting 0.6 noozle and 0.4 as temlate printer is not working lines are still set for 0.4

andygodber, 2024-07-09:
Problem is still there in 5.1

JLOFPV, 2024-07-12:
Hello. The truth is that it is very frustrating that you buy a printer like this and they do not provide the correct support for the latest software. I hope it doesn't take long to launch the profiles for 0.6 and 0.8 in K1C

andygodber, 2024-07-12:
> Hello. The truth is that it is very frustrating that you buy a printer like this and they do not provide the correct support for the latest software. I hope it doesn't take long to launch the profiles for 0.6 and 0.8 in K1C

I agree, but in the meantime, I copied/created the Orca 0.6 profile into Creality print - basically change the line widths to 0.62 for everything apart from sparse infill and inner wall to 0.65. Change extruder to 0.6. Change the extruder nozzle diameter to 0.6 and layer height limits range 0.12 to 0.48.

I did change the nozzle size in printer.cfg as well.

You may need to tune your filaments, but most defaults have worked well. I get more stringing though, so work to do. However where stringing wouldnt be 'needed' there isnt any.

r-carballo, 2024-10-08:
Issue persists in Ceality Print 5.1.4.10249 for the K1C.  
We see several other nozzle options for the K2 Plus, but nothing for the other "Flagship Series" printers.
Please include the same set of nozzle options for all K series printers.
Keep in mind, some of us want to print and have fun.  Your software should enable that to the fullest to help build brand loyalty.  Otherwise, what makes your software special to us?  It's not just its name, it's the feature set and EASE OF USE.

SWGC11, 2024-11-01:
SO ANNOYING!!!! So much wasted time. Creality print 5.1.4.10249 is showing every nozzle except for my o.4mm which was working perfectly previously. Have unisntalled/ reinstalled even previous versions of software and the K1 Max is updated and reset and still no 0,4 nozzle!!!! I showing all other machines except K1 Max in the "PROCESS" "GLOBAL" "DEFAULT CONFIGURATION" !!
pls help!!! aBSOLUTELY PATHETIC SOFTWARE AND NO INFORMATION OR CONTACT BASE WITH CREALITY . THIS COMPANY IS AN ABSOLUTE JOKE

closed 2025-04-26 by github-actions[bot]

SWGC11, 2025-04-27:
Creality print wORST CUSTOMER SERVICE EVER YOU CHUNTS!

Regards,

Suzy Weeks
M: 0466883977

IMPORTANT: This email and any attachments may contain legally privileged,
confidential or private information, and may be protected by copyright. You
may only use this information if you are the intended recipient(s) and if
you use it in an authorised way. This content is not to be shared by the
recipient(s) with a third party in any way without my written permission.
No other person is allowed to use, review, alter, transmit, disclose,
distribute, print or copy this email and any attachments without my prior
written consent.

If you are not the intended recipient(s) and the email was sent to you by
mistake, please notify the sender immediately by return email or phone,
destroy any hardcopies of this email and any attachments and delete it from
your system. Any legal privilege and confidentiality attached to this email
is not waived or destroyed by that mistake.