CrealityOfficial/CrealityPrint #122: Custom filament preferences are not saved.

alextyshchenko, 2024-05-17:
Hi, I can create new custom filament parameters, but I can use it only in one project.
When I restart Creality Print, I see my filaments as default (can't change them) with broken parameters.

For each project I should create my filament again and again.

[image]
[image]
[image]

P.S.: We have 2Gb in creality cloud. Please use that space for save all settings from Creality Print.

Regards,
Alex

whisky667, 2024-05-18:
I second this. It's essential to quicly import/export params. In a current state, the slicer is not usable as a tool.

danielbolognani, 2024-05-26:
Same problemas here! It is almost unusable if you dont use the default materials

Thyoity, 2024-06-04:
This needs attention, I lost all of my previous settings and wasted fillaments for material callibration.

lrovani, 2024-06-04:
Same problem, it's almost impossible to use. Everytime I create a new profile, the previous bugs and all configurations become weird. I was so happy with my K1 Max but this slicer has lots of problems. :(

naga-sk, 2024-06-14:
I emailed creating and they provided link to download v5.0.3 which addressed the issue with material management
https://github.com/CrealityOfficial/CrealityPrint/releases/download/v5.0.3/Creality_Print-v5.0.3.8736-win64-Release.exe

I tested it and I'm now able to see the custom materials listed separately and I'm able to edit / delete them

closed 2025-04-26 by github-actions[bot]