CrealityOfficial/CrealityPrint #128: Creality Print closes abnormaly on macOS

rxfxt, 2024-05-20:
Currently whenever I start Creality Print 5.0.2, I get the following error on macOS:

[image]

To reproduce:
1. Open Creality Print
2. Hit red close button in top left corner
3. Reopen Creality Print, notice error

ducklingasa, 2024-05-27:
Hi Creality,

I got the same error message almost every time on Windows as well, independently of how I close the program.

[image]

Reproducing steps are the same as for @rxfxt.

brannos, 2024-06-15:
my mac creality print app shows the splash screen then closes. Not error message

closed 2025-04-26 by github-actions[bot]