CrealityOfficial/CrealityPrint #140: Creality Print V5 for Mac - does not size to screen, UI goes blank

DAF54, 2024-05-25:
Creality Print V5 for Mac - does not size to screen, UI goes blank, Dragging to resize UI crashed, maximising UI to full screen and returning to previous state causes screen to go blanks White. On opening V5 on Mac does open to fill screen size, resizing manually causes blank screen and crash dump when closing App

The diagrams display frame (bottom right corner of UI) does not work at all, for parameters, illustrating complex parameters in an intuitive way. Not functioning at all on Mac version since release, also only functioning on Windows version.

Intermittent crashes while using Mac version 5 latest release, tried to upload crash dump error file was too large for your system to handle here..

DAF54, 2024-05-25:
[image]
[image]

LanceBV, 2024-05-25:
Same problem here for V 5.0.3 on Win. Doesnt crash but screen wont full size and mouse cursor is 1cm off the top of buttons.

soundtrax6, 2024-05-27:
Same problem here for V 5.0.3
Modellnamn:	iMac
  Modellidentifierare:	iMac19,2
  Processornamn:	Quad-Core Intel Core i3
  Processorhastighet:	3,6 GHz
  Antal processorer:	1
  Totalt antal kärnor:	4
  L2-cache (per kärna):	256 K
  L3-cache:	6 MB
  Minne:	8 GB
  Version av fast systemprogramvara:	2022.100.22.0.0
  OS-inläsarversion:	580~2215
  SMC-version (system):	2.47f3

geogeorge-nurt, 2024-06-28:
Same problem here for v5.0.3.8744-win64 on Win.10 wont full size and mouse cursor is 1cm off the top of buttons. i have tried everything.

DAF54, 2024-06-29:
When will this issue be resolved it still not working at all in Windows & Mac versions of V5.0.3.8744. - I really can not see any improvements in the Mac or Window versions both still have the same old UI issues, window scaling issues, general UI faults when zooming, mouse usability,  and scaling models, really how long before you address usability do we have to wait.. Q&A testing before releases seems totally lacking..

DAF54, 2024-07-05:
OMG.. Still not fixed this issue in V5.1.0.9484, how long before you will resolve these basic UI and usability issues, come on please fix these..

geogeorge-nurt, 2024-07-05:
Yes I totally agree. Come on Creality!!!

d4e5v8, 2024-07-18:
Hello Houston... or should I say Hello Wuhan (LOL)... same issue... actually worse with the latest version, it seems to start blinking if I switch to another app and come back, even without resizing...

DAF54, 2024-08-03:
Screen still goes blank when scaling application to fit screen OMG come on, please fix this issues in 5.1.2.9904 release

DAF54, 2024-08-03:
[image]

DAF54, 2024-08-03:
30 yrs working in software development company, I cannot believe your not able to resolve the most basic UI and general bugs, your Q&A is a waste of time... get the basics right please, this is so frustrating as an end user.. time and time again the same simple issues are not resolved in every successive release..

closed 2025-04-27 by github-actions[bot]