CrealityOfficial/CrealityPrint #146: Creality Print for linux

carlesoriol, 2024-05-27:
Creality print 5  release for linux is not available.

BeethovenBravo, 2024-05-27:
A linux version is better than wine.

carlesoriol, 2024-05-27:
> A linux version is better than wine.

Linux version 5 is not available.

carlesoriol, 2024-05-27:
> It works well ( including LAN printing ) under my Ubuntu 22.04 with wine 9.x

Nice trolling, but that's not linux version.

hagbardZ, 2024-05-29:
i kicked this peace of shit away, and installed orca ... this is way better on Linux. Sources are also available.
There is no reason to use this crap.

ronoaldo, 2024-05-31:
Running the slicer under wine is not ideal but it is not trolling. I'm using wine as well, and for me with K1 Max and local filament brands it is working quite well even without too much calibration needed to get decent prints.

There are also several issues already about Linux support: https://github.com/CrealityOfficial/CrealityPrint/issues?q=is%3Aissue+linux+

Creating more issues will just pollute the tracker... It is wiser to have a single point where we can keep informed instead of having several similar issues.

jaaxx, 2024-06-07:
I would like to confirm as of 5.0.3 release this issue is still not resolved.  Thanks.

carlesoriol, 2024-06-09:
Still no news on linux version?

carlesoriol, 2024-09-30:
5.1.2 Issue still in 5.1.2.

UnclePaulus, 2024-11-08:
I just tried the  Creality_Print-v5.1.6.10470-x86_64-Release.appimage on my K1 Max and it crashes almost immediately on my Debian 12 Bookworm platform.
Is there a solution to this?

UnclePaulus, 2024-11-10:
Sometimes the software starts correctly. Most of the times it doesn't. Being an old fool, I have been frantically clicking the right and left mouse-button during start up. And believe it or not... the software started...

UnclePaulus, 2024-11-12:
It is a matter of ssl indeed.
I found this post and it worked for me.
https://github.com/CrealityOfficial/CrealityPrint/issues/231
I didn't have to go further than 'sentence' 1897

ronoaldo, 2024-11-14:
> It is a matter of ssl indeed. I found this post and it worked for me. #231 I didn't have to go further than 'sentence' 1897

The fix to install the Ubuntu libssl1.1 package worked for me too! In my case, I had a different one, from Debian 11, laying around on my system. However, with the Ubuntu version it worked.

Curiously, based on the 5.0.3 cmake (only source code branch available), these files should be copied over to the .AppImage - not sure how the binaries are built but this seems to be either not working properly or missing from the current released version.

closed 2025-04-27 by github-actions[bot]