CrealityOfficial/CrealityPrint #156: No Way to Pan View on Laptop

raycardillo, 2024-05-31:
From what I understand, Cura and other slicers allow a shift right click to pan the view (move X Y in view instead of rotate). Some have reported that Creality Print supports this using the middle mouse button, but I'm using the software on a RTX 4090 laptop and do not have access to a middle mouse button on the touchpad. So this is very annoying.

I also use a 3d spacemouse but there is no support for that type of controller either. So the only way I can move around in the view is to zoom out and zoom in many times. It's a horrible user experience and a huge waste of time.

Please implement view panning using an alt button (shift or other key) in combination with right mouse click.

kilinccagatay, 2025-02-14:
Following as well!

andcam, 2025-04-12:
11 months later, still can't pan on a macbook with trackpad.

TheJadeGeek, 2025-05-09:
Ctrl right click on a Mac will allow you to pan, Not sure in windows which will help.