CrealityOfficial/CrealityPrint #158: Windows version. UI problem

pwitkowski1994, 2024-05-31:
Hi,

Problem is with
-size of ui window, cannot maximilize
-problem with cursor, it is offset against window

pwitkowski1994, 2024-05-31:
[image]

carlos-chierici, 2024-06-01:
Same problem here - using v5.0.3.

mattmaltarich, 2024-06-02:
Same issue on Windows.  Makes the software unusable. Your cursor has to be about half an inch above whatever you are trying to click to get it to activate.  Went back to the old version.  Looked cool, but not usable in this version.  Looking forward to an update.

SHD3R, 2024-06-06:
I have exactly the same problem on 5.0.0, 5.0.2 and 5.0.3

louisbotha, 2024-06-15:
Same issue, I have a monitor with a high aspect ratio. Make the app useless.

Faz3270, 2024-06-17:
As anyone found a fix for this issue on Windows 10? desktop pc and laptop both the same, I've tried a few things but its still the same.
come on Creality fix this please!

craig-childs, 2024-06-18:
Ditto, 
Windows 11, Creality Print v.5.0.0.8636
Screen resolution 1366 x 768, Samsung laptop integrated display.
Device name	Laptop-Samsung-NP300C
Processor	Intel(R) Core(TM) i5-3210M CPU @ 2.50GHz   2.50 GHz
Installed RAM	6.00 GB (5.79 GB usable)
System type	64-bit operating system, x64-based processor

Software unusable!

DoinStuff35, 2024-06-29:
Same hier / Win 10

SHD3R, 2024-06-29:
Still does not work with the latest release 5.0.3.8733 from July 24. This is how it looks on full screen. It is completely unusable.
[image]

DAF54, 2024-06-30:
The lower right corner preview window is totally useless in Mac and Windows versions, it is does preview at all, the window is cut off and hidden due to the bug ridden application scaling screen size and UI issues. Application scaling and sizing bug that has been an issue since the first release. running application full screen by dragging application window to size needed crashes the application and turns the entire screen white.. all bugs and issues still unresolved in V5.0.3.8744 Mac and Windows..

DAF54, 2024-06-30:
PLEASE ADDRESS THE BASIC UI AND SCALING ISSUES FOR THE APPLICATION WINDOW

GC1268, 2024-06-30:
Same problem, sort of
Version 5.0.3
Windows 11

I had the same issue on first installing, did some searching and found comments about changing display settings in the app properties.  went in had a look made some changes and the bottom right corner seems to be right??
However when I click the LAN printing button, that window doesn't load properly, it didn't from the off.  Changed any changes for the display in the app properties back to what they were originally and the bottom right corner stayed as image below.

[image]

[image]

orangeworx, 2024-07-08:
Ditto what everyone said. This started with 5 and every update since hasn't addressed the issue... always go about half an inch higher, unless it's for any pop up window.

carlos-chierici, 2024-07-16:
Creality launched Creality Print v5.1.1.9490 and this problem still remains...

orangeworx, 2024-07-16:
Updated, again, issue remains.
This is quite frustrating.

SHD3R, 2024-08-02:
Problem is finally fixed in the latest 5.1.2.9904 release - "Addressed an issue where mouse pointer offset in certain graphics card environments caused functionality buttons to be unclickable."

You can get it from CrealityCloud - https://www.crealitycloud.com/software-firmware/software/creality-print

edgedj, 2024-10-16:
Still a problem with V5.1.4.10249 for me on a fairly old Fujitsu tablet with an integrated Intel 620 graphics card. Ideally this would work as I do drawing on my more powerful machines and this sits next to my K1C sending the print files

twistedsanity, 2024-11-19:
Guess this issue may also be related to the scaling as I use a 7680x1440 display and the app opens full screen. When reducing to a more usable size the top row of buttons become so small they are almost unusable.
[image]

twistedsanity, 2024-12-17:
Just install v5.1.7.10514-win64-release and my scaling issues with the toolbar have been resolved. I wonder if the other items also a fixed but the same release.

closed 2025-04-27 by github-actions[bot]