CrealityOfficial/CrealityPrint #159: Closing V5.03 on Mac causes crasher error

DAF54, 2024-06-01:
Closing V5.03 on Mac causes crasher error, this has been an issue with V5 since release I have attached the full error dump report generated, hope this helps. There are inherent issues with the UI windows sizing and elements of the UI are not viewable. I have reported this also previously, look forward these issues being resolved

[Crealty Print 5.03 Crash Error Dump.pdf](https://github.com/user-attachments/files/15520042/Crealty.Print.5.03.Crash.Error.Dump.pdf)

DAF54, 2024-06-29:
When will this issue be resolved it still not working at all in Windows & Mac versions of V5.0.3.8744. - I really can not see any improvements in the Mac or Window versions both still have the same old UI issues, window scaling issues, general UI faults when zooming, mouse usability,  and scaling models, really how long before you address usability do we have to wait.. Q&A testing before releases seems totally lacking..

DAF54, 2024-07-05:
OMG.. Still not fixed this issue in V5.1.0.9484, how long before you will resolve these basic UI and usability issues, come on please fix these.. still same crash errors that I sent you two releases ago in this version ...

DAF54, 2024-07-20:
Still the same crash issues in V5.11 - How long is it going to take to get this sorted

DAF54, 2024-08-03:
Still same crash error on closing app in 5.1.2.9904 release

gcusick, 2024-08-23:
### _**Still there in 5.1.2**_

closed 2025-04-27 by github-actions[bot]