CrealityOfficial/CrealityPrint #161: Zoom control of Model in Mac version 5.03 - is uncontrollable and touchy, unusable.

DAF54, 2024-06-01:
Zoom control of Model in Mac version 5.03 - is uncontrollable and touchy, unusable.

DAF54, 2024-06-29:
When will this issue be resolved it still not working at all in Windows & Mac versions of V5.0.3.8744. - I really can not see any improvements in the Mac or Window versions both still have the same old UI issues, window scaling issues, general UI faults when zooming, mouse usability,  and scaling models, really how long before you address usability do we have to wait.. Q&A testing before releases seems totally lacking..

DAF54, 2024-07-05:
OMG.. Still not fixed this issue in V5.1.0.9484, how long before you will resolve these basic UI and usability issues, come on please fix these..

DAF54, 2024-08-03:
Still not resolved in 5.1.2.9904 release in 5.1.2.9904 release

DAF54, 2024-10-16:
still remains unresolved in Mac version... doubt this will ever be addressed

benclewett, 2024-11-17:
On my Mac this may be related to excessive CPU use, never sitting below 100%.  Therefore not enough processor time to smoothly move objects.  Which makes the program almost unsuable.

DAF54, 2024-12-18:
Still not resolved in V5.1.7.10514. - please put some effort into resolving this

closed 2025-04-27 by github-actions[bot]