CrealityOfficial/CrealityPrint #178: Usage with Ender 3 V3 SE

ELKER54, 2024-06-18:
Hi, I know V5 is still under development but apparently this should work with the Ender 3 V3 SE. 

However, when I try to slice and export the print it works, but the printer doesn't print properly and just shows 0's on estimated time, Filament used and layer height.
[image]

Not sure if I'm just being dumb or if a different setting is needed, I have tried on both the Ender 3 V3 and Ender 3 V3 KE profiles. 

Has anyone else had any luck?

beyondsoft978, 2024-07-04:
After 5.0, the orca kernel was used, so it does not support displaying machine preview time in the previous marlin firmware. I also encountered this issue when using ocra. In the future, it will be the world of kipper

beyondsoft978, 2024-07-12:
https://github.com/SoftFever/OrcaSlicer/issues/3546

user python script is Perfectly solve problems