CrealityOfficial/CrealityPrint #181: Hotkeys Not Working on Mac

Kvis-Dev, 2024-06-21:
Hotkeys in Creality Print do not function on macOS, making it difficult to use shortcuts for common actions like delete object or import model. This affects user productivity and needs to be addressed for better UX.

Its-The-Fuzz, 2024-06-24:
Hotkeys do work on my Mac, at least in the Prepare tab. In the Preview tab all menu commands are disabled, for some reason.

DAF54, 2024-06-30:
Same issues still in V5.0.3.8744. - I really can not see any improvements in the Mac or Window versions both still have the same old UI issues, window scaling issues, general UI faults when zooming, mouse usability,  and scaling models, really how long before you address usability do we have to wait.. Q&A testing before releases seems totally lacking..

closed 2025-04-27 by github-actions[bot]