CrealityOfficial/CrealityPrint #183: Linux AppImage for Review

eduncan911, 2024-06-29:
Need to evaluate for linux.

feiticeir0, 2024-07-01:
Creality doesn't care about it's Linux users. 
If you search the Issues, there are countless requests of Linux users such as myself, requesting for Creality to release, if not compiled, the source code (they are required by license to do so) so we can compile it. 

They still have yet to comment on those issues. 

They keep releasing version of Windows and/or MAC, but not Linux...

ernani, 2024-08-08:
I've tried manually building the linux version but it doesn't go through:

I get this error:
INFO      Run: conan create --profile linux -s build_type=Debug /tmp/tmphdpr50cp opensource/linux
ERROR: conanfile.py doesn't declare package name or version

feiticeir0, 2024-08-29:
> I've tried manually building the linux version but it doesn't go through:
> 
> I get this error: INFO Run: conan create --profile linux -s build_type=Debug /tmp/tmphdpr50cp opensource/linux ERROR: conanfile.py doesn't declare package name or version

I've tried to build the 5.0.3 version - the only one they have source code for - and I'm stuck at conan... It's a freaking shame.

TeamFahQ, 2024-09-02:
I've tried building 4.3.8, but fails, posted my issue her [#203](https://github.com/CrealityOfficial/CrealityPrint/issues/203)

closed 2025-04-27 by github-actions[bot]