CrealityOfficial/CrealityPrint #191: Creality Print 4.3.7.6627 does not slice

paganaye, 2024-07-21:
I am in Crealtiy Print 4.3.7.6627 with Linux Mint 21.3 
I loaded the Benchy3D.stl model I can see my 3D printer shown 
On the Printer window : Ender3 V3 KE 0.4 nozzle All settings are as default. 

When I click [Slice] on the bottom right button, I can see it changes mode and 3 buttons : [Lan Printing] , [Export to local], [Export to Cloud] These 3 buttons only show for less than a second though. As soon as the processing please wait... message disapear (roughtly one second), I am back to Prepare mode with a single [Slice] Button. Clicking on the Preview button in the middle top exhibit the same behaviour.
I have done a 12 seconds video https://www.youtube.com/watch?v=irPgjJhVt7E that shows the problem

I am totally new to this it might well be that my slicing settings are wrong. It is all set to defaults.
If some of the settings are wrong. I would be more happy to a dialog telling me "there is nothing to slice with details"

eduncan911, 2024-08-29:
they completely erased the git history of anything 4.x.

5.x is now a clone from Orca.  and they are not going to help with anything with 4.x.

Feel free to scream in the "there's no linux release of 5.x" issue here: https://github.com/CrealityOfficial/CrealityPrint/issues/183

oo-dev17, 2024-10-05:
Use 4.3.8 on Linux (that solved the slice issue for me on a german system)

closed 2025-04-27 by github-actions[bot]