CrealityOfficial/CrealityPrint #192: Creality print appimage silently exits

alsvartr, 2024-07-21:
Creality print appimage silently exits:

```
» appimage-run Downloads/Creality_Print-v4.3.7.6627-x86_64-Release.AppImage
Creality_Print-v4.3.7.6627-x86_64-Release.AppImage installed in /home/alsvartr/.cache/appimage-run/594b0bed46f1498f781f81755b0f2f2ec7d1c9e836fc367e57807a787dccebe3
» 
```

I'm using NixOS, Gnome on wayland.

basilverthorn, 2024-11-26:
Same here, I got a little further today, but still no go.

QFont::fromString: Invalid description 'Fira Sans,10,-1,5,400,0,0,0,0,0,0,0,0,0,0,1'
QFont::fromString: Invalid description 'FiraCode Nerd Font Mono,10,-1,5,400,0,0,0,0,0,0,0,0,0,0,1'
QFont::fromString: Invalid description 'Fira Sans,10,-1,5,400,0,0,0,0,0,0,0,0,0,0,1'
QFont::fromString: Invalid description 'Fira Sans,10,-1,5,400,0,0,0,0,0,0,0,0,0,0,1'
Initializing StaticPrintConfigsCXBinManager open file size [377984]
parse binary stl ...face 7558

parse binary stl success...

Computing bounding box... Done.
x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
Computing bounding box... Done.
x = -57.2188 .. 57.2205, y = -49.9992 .. 50.0008, z = -0.0766628 .. 228.967
Computing bounding box... Done.
x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
Computing bounding box... Done.
x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
X connection to :1 broken (explicit kill or server shutdown).

krom, 2024-12-06:
I have same issue on kde 6.1 with wayland
```
$ ./Creality_Print-v5.1.6.10470-x86_64-Release.appimage       

Initializing StaticPrintConfigsCXBinManager open file size [377984]
parse binary stl ...face 7558

parse binary stl success...

Computing bounding box... Done.
  x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
Computing bounding box... Done.
  x = -57.2188 .. 57.2205, y = -49.9992 .. 50.0008, z = -0.0766628 .. 228.967
Computing bounding box... Done.
  x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
Computing bounding box... Done.
  x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
X connection to :1 broken (explicit kill or server shutdown).
```

But, I have xwayland extension

closed 2025-04-27 by github-actions[bot]