CrealityOfficial/CrealityPrint #196: 5.0 version for m2 processor

hozdemir, 2024-07-23:
I have an m2 mac and I want to make a simple 3D print, should I use an orca slicer for this? I don't want to use orca slicer as it seems unnecessarily complex. Also, I cannot send the print to the printer. I have to export the gcode and upload it physically from USB because the upload section in the orca slicer does not work either. I downloaded and installed version 5.0 of creality slicer, but I cannot adjust the print speed of the printer. The speed section appears empty and the application constantly gives errors and closes. I wonder how people continue their work with such terrible software support? Should I definitely use Windows or Intel Mac?

krzysiekslimak, 2024-07-24:
Totaly agree, I have M1 mac but it is mandatory to release x86_64 architecture version. Intel version run on M1 or higher M proccesors consumes resources.

mattheworres, 2024-11-20:
There looks to be an ARM64 version now in the latest official release, FYI.

closed 2025-04-27 by github-actions[bot]