CrealityOfficial/CrealityPrint #200: CR10 SE print time and material length is coming correctly.

Narashiman, 2024-08-05:
When i use Creality Slicer 5.x version to slice for CR10SE, I am able to send and start printing the designs remotely. But many parameters such as layer height, Time duration, Material length are coming as 0 or not being decoded by printed properly. 

[image]

mattheworres, 2024-11-27:
I too am having this issue. I was able to send a few successful prints, but it seems the nozzle was auto-updated and something about it changed to where it won't work.

Creality Print v4 still works, and is how I've sliced my last few models, but I have need for the improved support process in v5 that I can't use at the moment. Will update if I've found anything.

mattheworres, 2024-11-28:
I have found that, at least so far (I'll hopefully remember to come back and edit this) by configuring Cura to use the CR10-SE configuration [according to this guide](https://github.com/Ultimaker/Cura/wiki/Adding-new-machine-profiles-to-Cura) (the files were also out of Github, though the [machine definition](https://github.com/Ultimaker/Cura/blob/0aec95089bfc9529b578f8f732c8d68f03f1e57c/resources/definitions/creality_cr10se.def.json) isn't found (at least seemingly) in the latest commits in Ultimaker Cura main branch. I also needed [this variant](https://github.com/Ultimaker/Cura/blob/0aec95089bfc9529b578f8f732c8d68f03f1e57c/resources/variants/creality/creality_cr10se_0.4.inst.cfg) added to Cura's directories before I started Cura - this removes the warning about being incompatible with the printer, and enables me to slice the model and output a `.gcode` file. Importing the gcode file in Creality Print v5 allows me to send the model to the printer, and even though the printer doesn't display the image (another configuration I might be able to remedy with tinkering) and thus print the model - at least I hope. so far it looks good. Obviously, make sure the definition matches your nozzle, mine still has the stock .4 nozzle so that's the variant I took. You could download some or all of them and place them in Cura's folder (on my Macbook, it was at `[Contents of Ultimaker Cura.app]/Resources/share/cura/resources/` and from there drop files in either the `definitions` or `variants` folders before starting Cura.