CrealityOfficial/CrealityPrint #203: compile release-v4.3.8

TeamFahQ, 2024-08-10:
I'm trying to compile v release-v4.3.8 in linux, but there are so many errors. 
The README states
cmake, git, python3.7+, conan 1.50, compiler toolchains should be installed
But it's not very specific as the what toolchains are to be used.
Thru trial by fire I found the following needed to be installed
Qt5Multimedia, ninja-build, qtmultimedia5-dev, Qt5Svg-dev, Qt5SerialPort-dev, qttools5-dev,Qt5WebSockets, linuxdeployqt
I had issues with linuxdeployqt. From what I found with an internet search linuxdeployqt-continuous-x86_64.AppImage should work, so I downloaded it. Tried to launch it, but it fails to launch on it's own.  Figured I'd let cmake try it. I couldn't figure out where to put it though, so I added the path manually to CMakeCache.txt. Now I'm getting the following errors, google isn't much help, and I'm kinda stuck

CMake Error at cmake/ConfigureTarget.cmake:212 (add_library):
  Target "qtuser_3d" links to target "Qt5::3DRender" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?
Call Stack (most recent call first):
  cmake/lib.cmake:44 (__add_real_target)
  qtuser_3d/CMakeLists.txt:21 (__add_common_library)

CMake Error at cmake/ConfigureTarget.cmake:212 (add_library):
  Target "qtuser_3d" links to target "Qt5::3DExtras" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?
Call Stack (most recent call first):
  cmake/lib.cmake:44 (__add_real_target)
  qtuser_3d/CMakeLists.txt:21 (__add_common_library)

CMake Error at cmake/ConfigureTarget.cmake:212 (add_library):
  Target "qtuser_3d" links to target "Qt5::3DRenderPrivate" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?
Call Stack (most recent call first):
  cmake/lib.cmake:44 (__add_real_target)
  qtuser_3d/CMakeLists.txt:21 (__add_common_library)

CMake Error at cmake/ConfigureTarget.cmake:212 (add_library):
  Target "shader_entity" links to target "Qt5::3DRender" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?
Call Stack (most recent call first):
  cmake/lib.cmake:44 (__add_real_target)
  shader_entity/CMakeLists.txt:21 (__add_common_library)

CMake Error at cmake/ConfigureTarget.cmake:212 (add_library):
  Target "shader_entity" links to target "Qt5::3DExtras" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?
Call Stack (most recent call first):
  cmake/lib.cmake:44 (__add_real_target)
  shader_entity/CMakeLists.txt:21 (__add_common_library)

CMake Error at cmake/ConfigureTarget.cmake:212 (add_library):
  Target "shader_entity" links to target "Qt5::3DRenderPrivate" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?
Call Stack (most recent call first):
  cmake/lib.cmake:44 (__add_real_target)
  shader_entity/CMakeLists.txt:21 (__add_common_library)

CMake Error at cmake/ConfigureTarget.cmake:212 (add_library):
  Target "shader_entity" links to target "Qt5::3DRender" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?
Call Stack (most recent call first):
  cmake/lib.cmake:44 (__add_real_target)
  shader_entity/CMakeLists.txt:21 (__add_common_library)

CMake Error at cmake/ConfigureTarget.cmake:212 (add_library):
  Target "shader_entity" links to target "Qt5::3DExtras" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?
Call Stack (most recent call first):
  cmake/lib.cmake:44 (__add_real_target)
[... 770 lines cut ...]
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?
Call Stack (most recent call first):
  Creative3D/CMakeLists.txt:90 (__add_real_target)

CMake Error at cmake/ConfigureTarget.cmake:202 (add_executable):
  Target "Creative3D" links to target "Qt5::3DRender" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?
Call Stack (most recent call first):
  Creative3D/CMakeLists.txt:90 (__add_real_target)

CMake Error at cmake/ConfigureTarget.cmake:202 (add_executable):
  Target "Creative3D" links to target "Qt5::3DExtras" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?
Call Stack (most recent call first):
  Creative3D/CMakeLists.txt:90 (__add_real_target)

CMake Error at cmake/ConfigureTarget.cmake:202 (add_executable):
  Target "Creative3D" links to target "Qt5::3DRenderPrivate" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?
Call Stack (most recent call first):
  Creative3D/CMakeLists.txt:90 (__add_real_target)

TeamFahQ, 2024-08-20:
What am I missing?

eduncan911, 2024-08-29:
4.x was forked from prusaslicer.
5.x is a new fork from Orca.

they completely erased, and erased the entire history of anything before 5.x this year, in 2024.

iow, you are not going to get any help with 4.x.

closed 2025-04-27 by github-actions[bot]