CrealityOfficial/CrealityPrint #216: Compiling for linux (almost there?)

Tercus, 2024-09-04:
Hello there,

I tried to compile CrealityPrint for linux and was getting a lot of errors. I was able to solve SOME of them, but so far have been unsuccessful. Still, I want to document, what I have done so far.

**What you need:**

- cmake
- python3
- conan (installed via pip3 -> it MUST be version 1.64.0, so use `pip install conan==1.64.0`)
  - conan binary must be usable as a console command. I had to manually add the binary path to my PATH with `PATH=$PATH:/home/<username>/.local/bin`, because of how pip installed the conan binary (I had to login again to make this available in my terminal)
- QT5 stuff
  - `sudo apt install qtbase5-dev libqt5serialport5-dev libqt5websockets5-dev libqt5svg5-dev qttools5-dev qtmultimedia5-dev qtdeclarative5-dev qt3d5-dev`
  - I also installed these, but that was during troubleshooting: "libqt5quick5 libqt5qml5 libqt53dcore5 libqt53drender5 libqt53dextras5 libqt53dinput5 libqt53dlogic5 libqt53dquick5"
  - this was a pain to figure all out, and even now I am not sure if this list is everything I installed
- some things I am not entirely sure, but that I installed because I was getting errors:
  - llvm (`apt install llvm-11-dev`)
  - openMP (`apt install libomp-dev`)
- linuxdeployqt
  - I actually installed it by compiling it (it's easy enough), but mainly because I failed at making it work otherwise. Maybe an alias in bashrc would have been enough, but that somehow did not work for me.

**What you do:**

- Clone the repository
- make sure to also download connected submodules `git submodule update --init`
- create a conan profile with `conan profile new --detect linux`
- it recommends to update the profile, so I did `conan profile update settings.compiler.libcxx=libstdc++11 linux`
- run `./cmake/ci/cmake.py -c -b -e --channel_name=opensource`
  - be sure that this run actually completes successfully. For quite some time I missed that I was having errors here

**Where it fails:**

Right now I am getting error messages, because it tries to move some files and lacks permission to do so. At the end I get this:

> CMake Error at cmake/FileUtil.cmake:409 (FILE):
  FILE COPY cannot copy file
  "/home/tobias/.conan/data/ovdbutil/0.0.3/opensource/linux/package/d8d8aba822aaa76849d2f1bafe4a2a62a9f83b74/lib/libovdbutil.so"
  to "/usr/local/lib//libovdbutil.so": Permission denied.
Call Stack (most recent call first):
  cmake/ConfigureTarget.cmake:661 (__copy_find_targets)
  cmake/cc.cmake:20 (__import_target)
  cmake/conan.cmake:114 (__test_import)
  cmake/conan/Find_conan_ovdbutil.cmake:7 (__conan_import)
  cmake/conan.cmake:26 (find_package)
  cmake/cc.cmake:6 (__conan_find)
  plugins/plugin_hollow/CMakeLists.txt:8 (__cc_find)
\
\
-- import ovdbutil success.
-- CONAN ****** End Find ovdbutil
-- add test __add_real_target plugin_hollow-----------------------> dll
-- target plugin_hollow -> public [] private [BasicKernel;ovdbutil;BasicKernel]
-- plugin_hollow CMAKE_CXX_FLAGS : -fPIC -shared
-- add test __add_real_target plugin_drill-----------------------> dll
-- target plugin_drill -> public [] private [BasicKernel]
-- plugin_drill CMAKE_CXX_FLAGS : -fPIC -shared
-- add test __add_real_target plugin_distance_measure-----------------------> dll
-- target plugin_distance_measure -> public [] private [BasicKernel]
-- plugin_distance_measure CMAKE_CXX_FLAGS : -fPIC -shared
-- add test __add_real_target plugin_color-----------------------> dll
-- target plugin_color -> public [] private [BasicKernel]
-- plugin_color CMAKE_CXX_FLAGS : -fPIC -shared
-- add test __add_real_target plugin_support_painting-----------------------> dll
-- target plugin_support_painting -> public [] private [BasicKernel]
-- plugin_support_painting CMAKE_CXX_FLAGS : -fPIC -shared
-- add test __add_real_target plugin_seam_painting-----------------------> dll
-- target plugin_seam_painting -> public [] private [BasicKernel]
-- plugin_seam_painting CMAKE_CXX_FLAGS : -fPIC -shared
-- add test __add_real_target plugin_adaptive_layer-----------------------> dll
-- target plugin_adaptive_layer -> public [] private [BasicKernel]
-- plugin_adaptive_layer CMAKE_CXX_FLAGS : -fPIC -shared
-- add test __add_real_target CrealityPrint-----------------------> exe
-- qml plugin CrealityUI : /home/tobias/Coding/repositories/CrealityPrint/plugins/CrealityUI/qmldir
-- target CrealityPrint -> public [] private [BasicKernel;cxkernel;Qt5::Svg;Qt5::WebSockets;Qt5::Xml;-pthread]
-- win deploy tool -> [WINDEPLOYQT_EXECUTABLE-NOTFOUND]
-- mac deploy tool -> [MACDEPLOYQT_EXECUTABLE-NOTFOUND]
-- linux deploy tool -> [/usr/lib/qt5/bin/linuxdeployqt]
-- CrealityPrint CMAKE_CXX_FLAGS : -fPIC  -permissive
-- add test __add_real_target unittesttool-----------------------> exe
-- target unittesttool -> public [] private [Qt5::Svg;Qt5::Core;Qt5::Quick;Qt5::Qml]
-- unittesttool CMAKE_CXX_FLAGS : -fPIC
-- Package Projects.
-- Package Projects
-- 0
-- Configuring incomplete, errors occurred!
ERROR     RUN FAILED.
********************************************

If anyone has any idea what I could try next, please let me know. I somehow do not want to run it as superuser to check if failing to move the files is the current cause for the failed run.

LumitoLuma, 2024-09-05:
As root,

```
# chown -R $USER:$USER /usr/local/lib
```

After CP is compiled,

```
# chown -R root:root /usr/local/lib
```

Tercus, 2024-09-05:
naaw...I don't think I want to do that on my PC. I'll try to recreate the conditions in a VM and try it there....

Tercus, 2024-09-06:
> As root,
> 
> ```
> # chown -R $USER:$USER /usr/local/lib
> ```
> 
> After CP is compiled,
> 
> ```
> # chown -R root:root /usr/local/lib
> ```

I setup a VM and got to the exact same problem. Then I did what you mentioned (after all the folder was almost empty anyway) and that actually helped me reach the next step.
I am now getting pretty much the same errors as mentioned in #203.

LumitoLuma, 2024-09-06:
Are you using the distribution's Qt release or the official one?

Haven't tried building 5.x on Linux, but for 4.x I needed to use both Qt's official builds (ie. 5.15.2) and CMake from Qt's Online Installer. Then, after adding Qt's CMake directory (which contains the cmake binary) in front of the PATH variable, worked for me.

Tercus, 2024-09-06:
I tried installing the cmake from Qt's official builds and it _somewhat_ changed the error message. Right now I still get the following error:

> -- Package Projects.
> -- Package Projects
> -- 0
> -- Configuring done (0.7s)
> CMake Error at cmake/ConfigureTarget.cmake:334 (target_link_libraries):
>   Target "qtuser_3d" links to:
> 
>     Qt5::3DRenderPrivate
> 
>   but the target was not found.  Possible reasons include:
> 
>     * There is a typo in the target name.
>     * A find_package call is missing for an IMPORTED target.
>     * An ALIAS target is missing.
> 
> Call Stack (most recent call first):
>   cmake/lib.cmake:44 (__add_real_target)
>   qtuser_3d/CMakeLists.txt:21 (__add_common_library)
> 
> 
> CMake Error at cmake/ConfigureTarget.cmake:334 (target_link_libraries):
>   Target "shader_entity" links to:
> 
>     Qt5::3DRenderPrivate
> 
>   but the target was not found.  Possible reasons include:
> 
>     * There is a typo in the target name.
>     * A find_package call is missing for an IMPORTED target.
>     * An ALIAS target is missing.
> 
> Call Stack (most recent call first):
>   cmake/lib.cmake:44 (__add_real_target)
>   shader_entity/CMakeLists.txt:21 (__add_common_library)
> 
> 
> -- Generating done (0.4s)
> CMake Generate step failed.  Build files cannot be regenerated correctly.
> ERROR     RUN FAILED.
> ********************************************

When I look online, I can barely find any info on this "Qt5::3DRenderPrivate". I then resorted to asking chatgpt and apparently this is a private function not meant to be used by developers directly (basically a system-function?). I just went ahead and removed those from the two CMakeLists.txt that had that reference.
I then got the following:

> -- Package Projects.
> -- Package Projects
> -- 0
> -- Configuring done (0.7s)
> -- Generating done (0.4s)
> -- Build files have been written to: /home/tobias/repos/CrealityPrint/linux-build/build
> INFO      RUN SUCCESS.
> ********************************************
> 
> shader binarization begin.
> project path: /home/<me>/repos/CrealityPrint/linux-build/build
> gl3 shader directory: /home/<me>/repos/CrealityPrint/cmake/ci/../../shader_entity/shaders/gl/3.3
> gles shader directory: /home/<me>/repos/CrealityPrint/cmake/ci/../../shader_entity/shaders/gles/3
> output: /home/<me>/repos/CrealityPrint/linux-build/build/shader_entity/GL.code
> shader binarization completed.

One step further, I guess? But what do I do now? I checked the `/linux-build/build/` folder, but could not find any binaries. Am I getting something wrong?

LumitoLuma, 2024-09-06:
Are you installing just Qt's cmake or Qt's cmake + Qt 5.15.2 from the Qt Online Installer (you need to toggle the "archived" checkbox to see the 5.15.2 build)? If you have both installed (Qt 5.15.2 with all of its extensions) and have uninstalled the respective system packages, try setting the `CMAKE_PREFIX_PATH` environment variable to point to the Qt 5.15.2 binaries folder (inside the `gcc_64` folder or something similar).

> One step further, I guess? But what do I do now? I checked the /linux-build/build/ folder, but could not find any binaries. Am I getting something wrong?

After building, you need to `cd` to the `/linux-build/build/` dir and run `cmake --build .`. If that command succeeds, you then should obtain the binaries.

yoprogramo, 2024-09-07:
Hello, I want to collaborate, but I will use docker to do so. Do you have some base Dockerfile or shoud I create a new one?

alexwbaule, 2024-09-11:
@Tercus  you got some errors on clipper2/0.0.1

``` C++

-- Build files have been written to: /home/alex/.conan/data/clipper2/0.0.1/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c
[ 25%] Building CXX object clipper2/CMakeFiles/clipper2.dir/src/clipper.engine.cpp.o
[ 50%] Building CXX object clipper2/CMakeFiles/clipper2.dir/src/clipper.rectclip.cpp.o
[ 75%] Building CXX object clipper2/CMakeFiles/clipper2.dir/src/clipper.offset.cpp.o
In file included from /home/alex/.conan/data/clipper2/0.0.1/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/clipper2/include/clipper2/clipper.h:16,
                 from /home/alex/.conan/data/clipper2/0.0.1/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/clipper2/src/clipper.rectclip.cpp:13:
/home/alex/.conan/data/clipper2/0.0.1/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/clipper2/include/clipper2/clipper.core.h: In member function ‘void Clipper2Lib::Point<T>::Init(T2, T2)’:
/home/alex/.conan/data/clipper2/0.0.1/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/clipper2/include/clipper2/clipper.core.h:98:20: warning: ‘if constexpr’ only available with ‘-std=c++17’ or ‘-std=gnu++17’ [-Wc++17-extensions]
   98 |                 if constexpr (std::numeric_limits<T>::is_integer &&
      |                    ^~~~~~~~~
/home/alex/.conan/data/clipper2/0.0.1/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/clipper2/include/clipper2/clipper.core.h: In function ‘Clipper2Lib::Rect<T1> Clipper2Lib::ScaleRect(const Rect<T2>&, double)’:
/home/alex/.conan/data/clipper2/0.0.1/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/clipper2/include/clipper2/clipper.core.h:433:12: warning: ‘if constexpr’ only available with ‘-std=c++17’ or ‘-std=gnu++17’ [-Wc++17-extensions]
  433 |         if constexpr (std::numeric_limits<T1>::is_integer &&
      |            ^~~~~~~~~
In file included from /home/alex/.conan/data/clipper2/0.0.1/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/clipper2/include/clipper2/clipper.h:16,
                 from /home/alex/.conan/data/clipper2/0.0.1/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/clipper2/src/clipper.offset.cpp:13:
/home/alex/.conan/data/clipper2/0.0.1/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/clipper2/include/clipper2/clipper.core.h: In member function ‘void Clipper2Lib::Point<T>::Init(T2, T2)’:
/home/alex/.conan/data/clipper2/0.0.1/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/clipper2/include/clipper2/clipper.core.h:98:20: warning: ‘if constexpr’ only available with ‘-std=c++17’ or ‘-std=gnu++17’ [-Wc++17-extensions]
   98 |                 if constexpr (std::numeric_limits<T>::is_integer &&
      |                    ^~~~~~~~~
/home/alex/.conan/data/clipper2/0.0.1/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/clipper2/include/clipper2/clipper.core.h: In function ‘Clipper2Lib::Rect<T1> Clipper2Lib::ScaleRect(const Rect<T2>&, double)’:
/home/alex/.conan/data/clipper2/0.0.1/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/clipper2/include/clipper2/clipper.core.h:433:12: warning: ‘if constexpr’ only available with ‘-std=c++17’ or ‘-std=gnu++17’ [-Wc++17-extensions]
  433 |         if constexpr (std::numeric_limits<T1>::is_integer &&
      |            ^~~~~~~~~
In file included from /home/alex/.conan/data/clipper2/0.0.1/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/clipper2/include/clipper2/clipper.engine.h:20,
                 from /home/alex/.conan/data/clipper2/0.0.1/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/clipper2/src/clipper.engine.cpp:17:
/home/alex/.conan/data/clipper2/0.0.1/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/clipper2/include/clipper2/clipper.core.h: In member function ‘void Clipper2Lib::Point<T>::Init(T2, T2)’:
/home/alex/.conan/data/clipper2/0.0.1/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/clipper2/include/clipper2/clipper.core.h:98:20: warning: ‘if constexpr’ only available with ‘-std=c++17’ or ‘-std=gnu++17’ [-Wc++17-extensions]
   98 |                 if constexpr (std::numeric_limits<T>::is_integer &&
      |                    ^~~~~~~~~
In file included from /home/alex/.conan/data/clipper2/0.0.1/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/clipper2/include/clipper2/clipper.h:17:
/home/alex/.conan/data/clipper2/0.0.1/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/clipper2/include/clipper2/clipper.engine.h: At global scope:
/home/alex/.conan/data/clipper2/0.0.1/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/clipper2/include/clipper2/clipper.engine.h:37:14: warning: elaborated-type-specifier for a scoped enum must not use the ‘class’ keyword
   37 |         enum class VertexFlags : uint32_t {
      |         ~~~~ ^~~~~
      |              -----
/home/alex/.conan/data/clipper2/0.0.1/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/clipper2/include/clipper2/clipper.engine.h:37:32: error: found ‘:’ in nested-name-specifier, expected ‘::’
   37 |         enum class VertexFlags : uint32_t {
      |                                ^
      |                                ::
/home/alex/.conan/data/clipper2/0.0.1/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/clipper2/include/clipper2/clipper.engine.h:37:20: error: ‘VertexFlags’ has not been declared
[... cut ...]
clipper2/0.0.1@opensource/linux: ERROR: Package '7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c' build failed
clipper2/0.0.1@opensource/linux: WARN: Build folder /home/alex/.conan/data/clipper2/0.0.1/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c
ERROR: clipper2/0.0.1@opensource/linux: Error in build() method, line 100
	cmake.build()
	ConanException: Error 2 while executing cmake --build '/home/alex/.conan/data/clipper2/0.0.1/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/.' '--' '-j24'
ERROR     RUN FAILED.
********************************************
```

allangj, 2024-09-23:
@yoprogramo were you able to create the dockerfile, I was going to do the same, create a docker container and play inside of it until find the correct recipe. If you have anything started or an image in a registry, please let me know to suffer a little less than you may have suffer :)

yoprogramo, 2024-09-23:
Hello @allangj , I have created a very initial dockerfile to work in, but I have found several problems trying to compile using this method. I clone the repo in the same directory as the Dockerfile / docker-compose and then execute this:

```
docker compose run --rm myapp bash
conan profile new linux --detect && conan profile update settings.compiler.libcxx=libstdc++11 linux
python ./cmake/ci/cmake.py -c -b -e --channel_name=opensource --cmake_args='-DCMAKE_BUILD_TYPE=Release'
```

Dockerfile
```
FROM ubuntu:22.04

# TZ y demás
RUN apt update -y
ENV TZ=Europe/Madrid
RUN ln -snf /usr/share/zoneinfo/$TZ /etc/localtime && echo $TZ > /etc/timezone
RUN DEBIAN_FRONTEND=noninteractive apt-get install -y locales
RUN sed -i -e 's/# es_ES.UTF-8 UTF-8/es_ES.UTF-8 UTF-8/' /etc/locale.gen && \
    dpkg-reconfigure --frontend=noninteractive locales && \
    update-locale LANG=es_ES.UTF-8
ENV LANG es_ES.UTF-8 
RUN apt-get install -y python3-pip python3-dev \
  && cd /usr/local/bin \
  && ln -s /usr/bin/python3 python \
  && pip3 install --upgrade pip

# Actualizamos el sistema y los paquetes
RUN apt-get update && apt-get install -y \
    build-essential \
    git \
    wget \
    unzip \
    zlib1g-dev \
    libicu-dev \
    libx11-dev \
    libxext-dev \
    libxcb1-dev \
    libgl1-mesa-dev \
    libglu1-mesa-dev \
    libgstreamer1.0-dev \
    libqt5serialport5-dev \
    libqt5websockets5-dev \
    libqt5svg5-dev \
    llvm-11-dev \
    libomp-dev \
    libomp-14-dev \
    qttools5-dev \
    qtmultimedia5-dev \
    qtdeclarative5-dev \
    qt3d5-dev \
    cmake
    # Agrega aquí cualquier otro paquete Qt que necesites

# Instalamos Conan
RUN pip3 install conan==1.64.0

# Configuramos el entorno de Conan
ENV CONAN_USER_HOME=/root/.conan

WORKDIR /app/CrealityPrint

CMD ["bash"]
```

docker-compose.yml:

```
version: '3'
services:
  myapp:
    build:
      context: .
      dockerfile: Dockerfile
    volumes:
      - ./CrealityPrint:/app/CrealityPrint
    command: bash

```

I'll try some other base docker (24.04) but I have received lot of errors after compiling long time.

allangj, 2024-09-24:
Hi, I compiled what was discussed in this thread into a repo that creates a docker image, bring up a container of that image and you can ssh to it. Initial setup was easy, however I found some things I had to manually do and will work later.
The repo: https://github.com/allangj/crealty_devenv
Things TODO:
1. I compiled manually qt5. I want to change that to use the online installer but still need to review the command line options to select the version 5 instead of default 6.
2. Linker try a -lzlib, but theres no libzlib, has it has a different name so I did a sym link. Need to review the correct fix.
3. There are some fixes to the code compiled in a patch in the files directory.
4. CMAKE_PREFIX_PATH doesnt like to play nice to me and has no effect when pointing to the QT5 directory and cmakes. I had to move libraries includes and plugins to standard locations to make it work, maybe with the online installer we can put that as default.

The compilation is extensive so probably you will want to leave the machine there and dont look at it because it will hang, or cry, or do something weird :rofl: 

Lastly, something to have in mind, is that the build binary uses dynamic libs to qt5, so you will need to have installed as well on the system you plan to run it. Im trying to review if with the ```make package``` it creates the appImage which is independent. Later will review if that work because so far I have a pretty invisible window that says crealityPrint5 (as I didnt had qt5 installed on the host machine)

feiticeir0, 2024-09-25:
I'm trying to compile this using Arch Linux and after created a Virtual Env and installed conan 1.64, I'm getting the following errors:

`********************************************
INFO      Run: conan create --profile linux -s build_type=Debug /tmp/tmp_kuox9v9 opensource/linux
WARN: *** Conan 1 is legacy and on a deprecation path ***
WARN: *** Please upgrade to Conan 2 ***
ERROR: Invalid setting '14' is not a valid 'settings.compiler.version' value.
Possible values are ['4.1', '4.4', '4.5', '4.6', '4.7', '4.8', '4.9', '5', '5.1', '5.2', '5.3', '5.4', '5.5', '6', '6.1', '6.2', '6.3', '6.4', '6.5', '7', '7.1', '7.2', '7.3', '7.4', '7.5', '8', '8.1', '8.2', '8.3', '8.4', '8.5', '9', '9.1', '9.2', '9.3', '9.4', '9.5', '10', '10.1', '10.2', '10.3', '10.4', '10.5', '11', '11.1', '11.2', '11.3', '11.4', '12', '12.1', '12.2', '12.3', '13', '13.1', '13.2']
Read "http://docs.conan.io/en/latest/faq/troubleshooting.html#error-invalid-setting"
ERROR     RUN FAILED.
********************************************
`

Oh, @CrealityTech , what about releasing a Linux version ?

allangj, 2024-09-26:
@feiticeir0 Initially I was following that approach, and creating the profile that way did not work for me neither.
I moved to the recommendation above of doing the following instead
```
conan profile new --detect linux && conan profile update settings.compiler.libcxx=libstdc++11 linux
``` 
That creates the profile for me.

I highly doubt there will be a linux official release soon, as you can from my previous comment, they even have basic compilation errors on the code. 
I already solved my previous error:
```
[qrc:/CrealityUI/qml/BasicMenuItemStyle.qml:87:9: QML Image: Cannot open: qrc:/UI/photo/submenu.png]
```
And now found more missing things. The steps I follow:
1. Setup Conan: ````conan profile new --detect linux && conan profile update settings.compiler.libcxx=libstdc++11 linux```
2. Compile from source code QT 5.15.2 (for some reason the online installation give me linking issues with some libs)
3. Setup CrealityPrint build dir: ```python3 ./cmake/ci/cmake.py -c -b -e --channel_name=opensource --cmake_args='-DCMAKE_BUILD_TYPE=Release -DOPENMP_ROOT=/usr/lib/llvm-11 -DCMAKE_PREFIX_PATH=<QT installation path>'```
4. Mode to build dir: ```cd linux-build/build/```
5. Build and install (it install things in a dir inside build): ```make install -j8```
6. Package to create a AppImage: ```make package -j8```

With that you will get an AppImage. It gives the error I mention before ```QML Image: Cannot open: qrc:/UI/photo/submenu.png]```
Which is because it cannot find the qml directory and it is not package in the appImage.
You can direct to the ones of your Qt directory: ```export QML2_IMPORT_PATH=<QT PATH>/Qt/5.15.2/gcc_64/qml```

Thats my current status, I do see the login and it tries to go up, but a  with a symbol error on some QT packages are preventing me to advance more. If all that I mention is happening then I doubt their CIT is running for linux, at least not with the released code

feiticeir0, 2024-09-26:
Thank you @allangj , but no cigar. 

I'm getting errors with trimesh... :
`-- Build files have been written to: /home/feiticeir0/.conan/data/trimesh2/0.0.3/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c
[  5%] Building CXX object trimesh2/CMakeFiles/trimesh2.dir/base/KDtree.cc.o
[ 11%] Building CXX object trimesh2/CMakeFiles/trimesh2.dir/base/TriMesh_connectivity.cc.o
[ 16%] Building CXX object trimesh2/CMakeFiles/trimesh2.dir/base/TriMesh_io.cc.o
[ 22%] Building CXX object trimesh2/CMakeFiles/trimesh2.dir/base/TriMesh_pointareas.cc.o
[ 27%] Building CXX object trimesh2/CMakeFiles/trimesh2.dir/base/Vec3Utils.cc.o
[ 33%] Building CXX object trimesh2/CMakeFiles/trimesh2.dir/base/TriMesh_tstrips.cc.o
[ 38%] Building CXX object trimesh2/CMakeFiles/trimesh2.dir/base/filter.cc.o
[ 44%] Building CXX object trimesh2/CMakeFiles/trimesh2.dir/base/TriMesh_bounding.cc.o
[ 50%] Building CXX object trimesh2/CMakeFiles/trimesh2.dir/base/TriMesh_curvature.cc.o
[ 55%] Building CXX object trimesh2/CMakeFiles/trimesh2.dir/base/TriMesh_grid.cc.o
[ 61%] Building CXX object trimesh2/CMakeFiles/trimesh2.dir/base/TriMesh_normals.cc.o
[ 72%] Building CXX object trimesh2/CMakeFiles/trimesh2.dir/base/remove.cc.o
[ 72%] Building CXX object trimesh2/CMakeFiles/trimesh2.dir/base/TriMesh_stats.cc.o
[ 77%] Building CXX object trimesh2/CMakeFiles/trimesh2.dir/base/reorder_verts.cc.o
[ 83%] Building CXX object trimesh2/CMakeFiles/trimesh2.dir/extend/quaternion.cpp.o
[ 88%] Building CXX object trimesh2/CMakeFiles/trimesh2.dir/extend/extendio.cpp.o
[ 94%] Building CXX object trimesh2/CMakeFiles/trimesh2.dir/extend/vertextransform.cpp.o
/home/feiticeir0/.conan/data/trimesh2/0.0.3/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/trimesh2/base/TriMesh_io.cc: In function ‘bool trimesh::IsBinarySTL(FILE*, unsigned int)’:
/home/feiticeir0/.conan/data/trimesh2/0.0.3/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/trimesh2/base/TriMesh_io.cc:1062:9: error: ‘uint32_t’ was not declared in this scope
 1062 |         uint32_t faceCount(0);
      |         ^~~~~~~~
/home/feiticeir0/.conan/data/trimesh2/0.0.3/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/trimesh2/base/TriMesh_io.cc:21:1: note: ‘uint32_t’ is defined in header ‘<cstdint>’; this is probably fixable by adding ‘#include <cstdint>’
   20 | #include "ccglobal/log.h"
  +++ |+#include <cstdint>
   21 | 
/home/feiticeir0/.conan/data/trimesh2/0.0.3/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/trimesh2/base/TriMesh_io.cc:1063:16: error: ‘faceCount’ was not declared in this scope
 1063 |         fread(&faceCount, sizeof(uint32_t), 1, f);
      |                ^~~~~~~~~
/home/feiticeir0/.conan/data/trimesh2/0.0.3/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/trimesh2/base/TriMesh_io.cc:1064:15: error: ‘uint32_t’ does not name a type
 1064 |         const uint32_t expectedBinaryFileSize = faceCount * 50 + 84;
      |               ^~~~~~~~
/home/feiticeir0/.conan/data/trimesh2/0.0.3/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/trimesh2/base/TriMesh_io.cc:1064:15: note: ‘uint32_t’ is defined in header ‘<cstdint>’; this is probably fixable by adding ‘#include <cstdint>’
/home/feiticeir0/.conan/data/trimesh2/0.0.3/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/trimesh2/base/TriMesh_io.cc:1066:16: error: ‘expectedBinaryFileSize’ was not declared in this scope
 1066 |         return expectedBinaryFileSize == fileSize;
      |                ^~~~~~~~~~~~~~~~~~~~~~
/home/feiticeir0/.conan/data/trimesh2/0.0.3/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/trimesh2/base/TriMesh_io.cc: In function ‘bool trimesh::write_obj(TriMesh*, FILE*, bool)’:
/home/feiticeir0/.conan/data/trimesh2/0.0.3/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/trimesh2/base/TriMesh_io.cc:2194:24: warning: ISO C++ forbids converting a string constant to ‘char*’ [-Wwrite-strings]
 2194 |         char* vtHead = "vt ";
      |                        ^~~~~
/home/feiticeir0/.conan/data/trimesh2/0.0.3/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/trimesh2/base/TriMesh_io.cc:2200:24: warning: ISO C++ forbids converting a string constant to ‘char*’ [-Wwrite-strings]
 2200 |         char* vnHead = "vn ";
      |                        ^~~~~
/home/feiticeir0/.conan/data/trimesh2/0.0.3/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/trimesh2/base/TriMesh_io.cc:2252:23: warning: ISO C++ forbids converting a string constant to ‘char*’ [-Wwrite-strings]
 2252 |         char* fHead = "f ";
      |                       ^~~~
/home/feiticeir0/.conan/data/trimesh2/0.0.3/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/trimesh2/base/Vec3Utils.cc: In function ‘float trimesh::vv_angle(const vec3&, const vec3&)’:
/home/feiticeir0/.conan/data/trimesh2/0.0.3/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/trimesh2/base/Vec3Utils.cc:13:27: error: ‘INT8_MIN’ was not declared in this scope
   13 |         if (denominator < INT8_MIN)
      |                           ^~~~~~~~
make[2]: *** [trimesh2/CMakeFiles/trimesh2.dir/build.make:216: trimesh2/CMakeFiles/trimesh2.dir/base/Vec3Utils.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [trimesh2/CMakeFiles/trimesh2.dir/build.make:146: trimesh2/CMakeFiles/trimesh2.dir/base/TriMesh_io.cc.o] Error 1
[... cut ...]
trimesh2/0.0.3@opensource/linux: ERROR: Package '7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c' build failed
trimesh2/0.0.3@opensource/linux: WARN: Build folder /home/feiticeir0/.conan/data/trimesh2/0.0.3/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c
ERROR: trimesh2/0.0.3@opensource/linux: Error in build() method, line 100
	cmake.build()
	ConanException: Error 2 while executing cmake --build '/home/feiticeir0/.conan/data/trimesh2/0.0.3/opensource/linux/build/7b80bc46b44f3cd3a482e5e5c0be38a03f2bc84c/.' '--' '-j24'
ERROR     RUN FAILED.
`

ssouthurst, 2024-10-07:
I've been trying to put the project together in Visual Studio - because Conan wouldn't work for me (never used it - spent ages trying to figure out why the batch couldn't find "conan").  In doing this I've found that while Creality have a version of Trimesh2 available in their GitHub - it is not the Trimesh they use in the CrealityPrint repo.

If you look at the conan recipes the references are largely sourced from a private IP (e.g. "http://172.20.180.12:8050/yanfa4/thirdparty/trimesh2").

When you look through the code that references trimesh - it expects some properties that don't exist.  e.g. textureIDs

https://github.com/CrealityOfficial/CrealityPrint/blob/7f1123cda2b85b2da6e877b61ed979c3cc7ebaa7/cxkernel/cxkernel/data/trimeshutils.cpp#L110

Personally I'm going to see if I can get Orca working with my Ender-3 Max Neo.

feiticeir0, 2024-10-07:
Since @CrealityTech  won't give us an answer - never have they answer once  - on all the Linux questions - I'm going to find alternatives... 
Since Creality Slicer is based on Orca, maybe Orca will work fine.  Will install the Creality Slicer in my windows, export the profiles and import them all in Orca... At least I'll have an up to date slicer - and if I like it, will stick to it. 
Will try to sell my K1C and buy a Bambu.. At least their slicer works on Linux !

ernani, 2024-10-07:
I am also switching my K1 Max to a Bambulab X1 Carbon due to these inconsistencies. Aside from having to replace 2 times the hotend and now upgraded to the new hotend, K1 Max is still beta when compared to Bambulab's product lines. I was hoping for the launch of K2 Plus but that will be alpha for a while as well, and, loads of anticipated headaches for sure.

ssouthurst, 2024-10-07:
Re: trimesh - I found the modified version - it's in the 2.0.0 branch (rather than the default branch)... :|

JaneSmith, 2024-10-23:
Has anyone managed to get a Linux build of this yet? I have a K2 Plus coming soon... would be great to be able to actually run the software for it...

BeethovenBravo, 2024-11-07:
> Has anyone managed to get a Linux build of this yet? I have a K2 Plus coming soon... would be great to be able to actually run the software for it...

No, currently, the only way to use Creality Print 5 on Linux is through Wine.

closed 2025-04-27 by github-actions[bot]