CrealityOfficial/CrealityPrint #218: creality print on m2 mac

hozdemir, 2024-09-10:
I've been waiting for a build for apple silicon for months.

I'm trying to use the app on my M2 mac. It works terribly with rosetta.

After using the app for a while and then quitting, it freezes. I have to kill and restart every time I want to use it again. I don't have another computer. I don't use Windows. Why did you stop looking into these issues? When are you going to create a build specifically for apple silicon processors?

gcusick, 2024-09-12:
Don’t hold your breath!  This has been open for months.

TheNightmanCodeth, 2024-10-27:
Looks like you can compile locally with `brew install conan@1`, then running the build (`python3 ./cmake/ci/build.py`) to generate the Conan profile and then `cp ~/.conan/profiles/default ~/.conan/profiles/mac-arm64` and running the build again. 

Unfortunately their copy of `boost_static` is not building. It looks like it fails during linking. Updating the dependency _might_ be all it takes to fix it but I have no idea how Conan works so I'm just gonna drop this here in the hopes someone else stumbles across this and can give it a shot

Its-The-Fuzz, 2024-11-07:
I've submitted a number of bug reports through several channels, and it's become clear that Creality simply does not read users' feedback. They ask for it, often, but they don't read it. So it's safe to assume that certain bugs will never be fixed.

dwaite, 2024-11-18:
There is an apple silicon release here (under releases) as of 5.1.6

gcusick, 2025-02-17:
###Still waiting!

Its-The-Fuzz, 2025-02-17:
They never even got all the functionality working in version 5 for Mac, so I'm not holding my breath for version 6.

wen422, 2025-05-19:
Creality bot: this issue is stale because it has been open for 90 days with no activity.

closed 2025-05-26 by github-actions[bot]