CrealityOfficial/CrealityPrint #226: Window/Screen Flickers constantly

jb510, 2024-10-27:
When trying CP V5.1.4.10249 the CP5 window will constantly flicker and runs very slow like errors are occurring constantly in the background.  

Note, the same thing happened the first time I tried to use an earlier version of CP5, like now I just gave up and went back to Orca. 
 I normally use Orca (2.1 and 2.2.x) and it runs fine, but again just wanted to try CP5.

I can successfully slice and send it to my printer with CP5, but it's visually painful to do so with the constant flickering.  

https://github.com/user-attachments/assets/834a0537-44ed-45e2-bad0-5ad4d295860a

https://github.com/user-attachments/assets/7a9eec83-6399-49eb-9a31-efb2c666ae82

https://github.com/user-attachments/assets/1d8a1577-6b48-4bcf-af46-aa5184d96075

I have no idea what is causing it, but it's persistent and ONLY affects the CP5 window.  Moving that window to only occupy part of the screen, the rest of the screen is stable but the window still flickers.  Window even flickers when it's a background window.  

Something is weirdly refreshing that window.  I couldn't find any logs for CP5 but I'm happy to share them if you tell me where they are.

Details:
2019 16" MacBook Pro MacOS 15.0.1 (also happened on 15.0.0)
Creality Print V5.1.4.10249

https://github.com/user-attachments/assets/6726a746-e8c7-4033-95a7-69fb9341177f

https://github.com/user-attachments/assets/5854efa3-66eb-4e51-bebc-73cafb42c07c

https://github.com/user-attachments/assets/47b24b42-5ef2-4f57-a06b-4d366a5d7842

https://github.com/user-attachments/assets/23d2c321-a6da-4e6c-b670-bd4414bc747d

jb510, 2024-10-31:
Public reports of the same are months old. https://www.reddit.com/r/Creality/comments/1dd2qux/creality_print_v5_flickering_non_stop/

benclewett, 2024-11-17:
This is also happening on the MacOS version V5.1.4.

mattheworres, 2024-11-20:
It happens with both v4 as well as v5. I tried dropping the refresh rate of the screen down to 60hz but it did not resolve the issue. Still seeing it on Sequoia 15.1 on an M1 chip.

janbraiins, 2024-12-02:
Can you actually fix this problem? This makes the software totally unusable on Mac

mattheworres, 2024-12-16:
I'd suggest using a slicer like Cura or PrusaSlicer well configured for your printer, and use your printer that way.

jb510, 2025-01-02:
Still happening in Creality Print 6.0

closed 2025-04-27 by github-actions[bot]