CrealityOfficial/CrealityPrint #231: Linux Appimage 5.1.6 not working un ubuntu

carlesoriol, 2024-11-08:
Release Creality_Print-v5.1.6.10470-x86_64-Release.AppImage crashes on ubuntu after start.

./Creality_Print-v5.1.6.10470-x86_64-Release.AppImage 
QSocketNotifier: Can only be used with threads started with QThread
qt.network.ssl: QSslSocket: cannot resolve EVP_PKEY_base_id
qt.network.ssl: QSslSocket: cannot resolve SSL_get_peer_certificate

wbvczar, 2024-11-09:
Same experience for me.  Ubuntu 24.04

BB31420, 2024-11-09:
Downgraded from SSL 3.0 to 1.1. App loaded, asked to select language but now crashes at open with

XIO:  fatal IO error 2 (No such file or directory) on X server ":0.0"
      after 742 requests (742 known processed) with 0 events remaining.
Segmentation fault (core dumped)

BB31420, 2024-11-09:
Running using:

`__NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=nvidia ./Creality_Print-v5.1.6.10470-x86_64-Release.AppImage`

BB31420, 2024-11-10:
1882  cd Downloads/
 1883  chmod +x Creality_Print-v5.1.6.10470-x86_64-Release.AppImage 
 1884  ./Creality_Print-v5.1.6.10470-x86_64-Release.AppImage 
 1885  sudo apt-get update
 1886  sudo apt-get install libssl1.1
 1887  sudo apt-get install libssl3
 1888  sudo apt-get install libqt5network5
 1889  ./Creality_Print-v5.1.6.10470-x86_64-Release.AppImage 
 1890  ./Creality_Print-v5.1.6.10470-x86_64-Release.AppImage  --appimage-extract
 1891  cd squashfs-root/
 1892  ./AppRun 
 1893  openssl version
 1894  wget http://nz2.archive.ubuntu.com/ubuntu/pool/main/o/openssl/libssl1.1_1.1.1f-1ubuntu2_amd64.deb
 1895  sudo dpkg -i libssl1.1_1.1.1f-1ubuntu2_amd64.deb
 1896  sudo apt --fix-broken install
 1897  ./AppRun 
 1898  cd Downloads/
 1899  ./Creality_Print-v5.1.6.10470-x86_64-Release.AppImage 
 1900  sudo apt-get install     libqt5gui5     libqt5widgets5     libqt5opengl5     libqt5dbus5     libx11-xcb1     libxcb-xinerama0     libxcb-icccm4     libxcb-image0     libxcb-keysyms1     libxcb-randr0     libxcb-render-util0
 1901  QT_DEBUG_PLUGINS=1 ./Creality_Print-v5.1.6.10470-x86_64-Release.AppImage
 1902  DISPLAY=:0 ./Creality_Print-v5.1.6.10470-x86_64-Release.AppImage
 1903  LIBGL_ALWAYS_SOFTWARE=1 ./Creality_Print-v5.1.6.10470-x86_64-Release.AppImage
 1904  MESA_GL_VERSION_OVERRIDE=3.3 ./Creality_Print-v5.1.6.10470-x86_64-Release.AppImage
 1905  X -version
 1906  sudo X -version
 1907  xhost +local:
 1908  QT_X11_NO_MITSHM=1 LIBGL_DRI3_DISABLE=1 ./Creality_Print-v5.1.6.10470-x86_64-Release.AppImage
 1909  qmake --version
 1910  sudo apt-get install libx11-dev libxext-dev libxrender-dev libxinerama-dev libxi-dev libxrandr-dev libxcursor-dev libxtst-dev
 1911  sudo apt install qt5-default qt5-qmake
 1912  glxinfo | grep "OpenGL vendor"
 1913  sudo apt install     qt5-default     libqt5core5a     libqt5gui5     libqt5widgets5     libqt5opengl5     libqt5network5
 1914  ./Creality_Print-v5.1.6.10470-x86_64-Release.AppImage 
 1915  export QT_PLUGIN_PATH=/usr/lib/x86_64-linux-gnu/qt5/plugins
 1916  export QT_QPA_PLATFORM_PLUGIN_PATH=/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms
 1917  ./Creality_Print-v5.1.6.10470-x86_64-Release.AppImage
 1918  sudo apt install     mesa-common-dev     libglu1-mesa-dev     libglew-dev     freeglut3-dev
 1919  lspci | grep -i vga
 1920  glxinfo | grep "OpenGL renderer"
 1921  export QT_PLUGIN_PATH=/usr/lib/x86_64-linux-gnu/qt5/plugins
 1922  export QT_QPA_PLATFORM_PLUGIN_PATH=/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms
 1923  export QT_DEBUG_PLUGINS=1
 1924  export QT_LOGGING_RULES="qt.qpa.*=true"
 1925  SEGFAULT_SIGNALS="all" ./Creality_Print-v5.1.6.10470-x86_64-Release.AppImage
 1926  export QT_PLUGIN_PATH=/usr/lib/x86_64-linux-gnu/qt5/plugins
 1927  export QT_QPA_PLATFORM_PLUGIN_PATH=/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms
 1928  export __GL_THREADED_OPTIMIZATIONS=1
 1929  export __GL_SYNC_TO_VBLANK=0
 1930  export LIBGL_ALWAYS_INDIRECT=1
 1931  ./Creality_Print-v5.1.6.10470-x86_64-Release.AppImage
 1932  export QT_PLUGIN_PATH=/usr/lib/x86_64-linux-gnu/qt5/plugins
 1933  export QT_QPA_PLATFORM_PLUGIN_PATH=/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms
 1934  export __GL_SHADER_DISK_CACHE=0
 1935  export MESA_GL_VERSION_OVERRIDE=4.3
 1936  ./Creality_Print-v5.1.6.10470-x86_64-Release.AppImage
 1937  sudo apt install nvidia-opengl-dev
 1938  sudo apt install nvidia-utils-dev
 1939  export QT_PLUGIN_PATH=/usr/lib/x86_64-linux-gnu/qt5/plugins
 1940  export QT_QPA_PLATFORM_PLUGIN_PATH=/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms
 1941  export __GL_SYNC_TO_VBLANK=0
 1942  export __GL_SHADER_DISK_CACHE=0
 1943  export LIBGL_ALWAYS_SOFTWARE=0
 1944  export MESA_GL_VERSION_OVERRIDE=3.3
 1945  export QT_OPENGL=desktop
 1946  ./Creality_Print-v5.1.6.10470-x86_64-Release.AppImage
 1947  glxinfo | grep "OpenGL version"
 1948  export QT_PLUGIN_PATH=/usr/lib/x86_64-linux-gnu/qt5/plugins
 1949  export QT_QPA_PLATFORM_PLUGIN_PATH=/usr/lib/x86_64-linux-gnu/qt5/plugins/platforms
 1950  export MESA_GL_VERSION_OVERRIDE=4.6
 1951  export __GL_SYNC_TO_VBLANK=0
[... 23 lines cut ...]
 1975  cd Downloads/
 1976  ./Creality_Print-v5.1.6.10470-x86_64-Release.AppImage 
 1977  cat /var/log/Xorg.0.log
 1978  grep "^\(EE\)" /var/log/Xorg.0.log
 1979  grep "^\(WW\)" /var/log/Xorg.0.log
 1980  journalctl -xe
 1981  echo $DISPLAY
 1982  who
 1983  sudo apt-get install -y libglu1-mesa libxcb-xinerama0
 1984  ./Creality_Print-v5.1.6.10470-x86_64-Release.AppImage 
 1985  sudo apt install -y libglx-mesa0 libgl1-mesa-dri libgl1-mesa-glx
 1986  E: Package 'libgl1-mesa-glx' has no installation candidate
 1987  sudo apt install -y libgl1-mesa-dev mesa-common-dev
 1988  lspci -k | grep -EA3 'VGA|3D|Display'
 1989  glxinfo | grep "OpenGL version"
 1990  sudo apt install mesa-utils
 1991  __NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=nvidia ./Creality_Print-v5.1.6.10470-x86_64-Release.AppImage
 1992  echo '[Desktop Entry]
 1993  Name=Creality Print
 1994  Comment=Creality 3D Slicer
 1995  Exec=env __NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=nvidia '"$HOME/Downloads/Creality_Print-v5.1.6.10470-x86_64-Release.AppImage"'
 1996  Icon=creality
 1997  Terminal=false
 1998  Type=Application
 1999  Categories=Graphics;3DGraphics;Engineering;' > ~/.local/share/applications/creality-slicer.desktop

AtomicRobotMan0101, 2024-11-10:
UGH. That was a waste of many hours.

Chasing THAT old ubuntu libssl1.1 down and trying to hammer it into place. 

For others, I wouldn't even bother, lets wait until the devs fix their library not to rely on ancient libraries and simply update it to 3+.

BB31420, 2024-11-10:
[image]

I tried to install this in a fresh 24.04 VM in order to document the process but there are additional issues that arise when trying to run in a virtual space. Here is the screenshot from my desktop build.

JaneSmith, 2024-11-11:
> Release Creality_Print-v5.1.6.10470-x86_64-Release.AppImage crashes on ubuntu after start.
>
> ./Creality_Print-v5.1.6.10470-x86_64-Release.AppImage QSocketNotifier: Can only be used with threads [...]

Same issue here, trying to run it inside a Fedora Distrobox container.

ygandahvn, 2024-11-12:
Until they fix it, you can do
`wget http://nz2.archive.ubuntu.com/ubuntu/pool/main/o/openssl/libssl1.1_1.1.1f-1ubuntu2_amd64.deb`
`sudo dpkg -i libssl1.1_1.1.1f-1ubuntu2_amd64.deb`
`export PATH=/opt/openssl-1.1/bin:$PATH`
`export LD_LIBRARY_PATH=/opt/openssl-1.1/lib:$LD_LIBRARY_PATH`

And run the appimage file

feiticeir0, 2024-11-13:
> Running using this after installing ssl1.1:
> 
> ```
> __NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=nvidia ./Creality_Print-v5.1.6.10470-x86_64-Release.AppImage
> ```
> 
> This worked with my nvidia gpu x11 setup. It might work by default with wayland or x11/amd?
> 
> Make creality.sh to avoid having to always run with the flag.
> 
> ```
> #!/bin/bash
> __NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=nvidia ./Creality_Print-v5.1.6.10470-x86_64-Release.AppImage
> ```
> 
> ```
> chmod +x creality.sh
> ```
> 
> Or do this and make a desktop shortcut/app entry
> 
> ```
> echo '[Desktop Entry]
> Name=Creality Print
> Comment=Creality 3D Slicer
> Exec=env __NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=nvidia '"$HOME/Downloads/Creality_Print-v5.1.6.10470-x86_64-Release.AppImage"'
> Icon=creality
> Terminal=false
> Type=Application
> Categories=Graphics;3DGraphics;Engineering;' > ~/.local/share/applications/creality-slicer.desktop
> ```
> 
> Change the icon here, then find it in the menu search to make a desktop shortcut.
> 
> ```
> cd ~/.local/share/applications/
> ```

For those having this also on Arch Linux, this solution works.

guarnacciaa, 2024-11-14:
> Running on 24.04, x11, and nvidia gpu with the default openssl 3.0.13 and the following flag:
> 
> ```
> __NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=nvidia ./Creality_Print-v5.1.6.10470-x86_64-Release.AppImage
> ```
> 
> Make creality.sh to avoid having to always run with the flag.
> 
> ```
> #!/bin/bash
> __NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=nvidia ./Creality_Print-v5.1.6.10470-x86_64-Release.AppImage
> ```
> 
> ```
> chmod +x creality.sh
> ```
> 
> Or do this and make a desktop shortcut/app entry
> 
> ```
> echo '[Desktop Entry]
> Name=Creality Print
> Comment=Creality 3D Slicer
> Exec=env __NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=nvidia '"$HOME/Downloads/Creality_Print-v5.1.6.10470-x86_64-Release.AppImage"'
> Icon=creality
> Terminal=false
> Type=Application
> Categories=Graphics;3DGraphics;Engineering;' > ~/.local/share/applications/creality-slicer.desktop
> ```
> 
> Change the icon here, then find it in the menu search to make a desktop shortcut.
> 
> ```
> cd ~/.local/share/applications/
> ```

This worked for me on Ubuntu 24.04

wbvczar, 2024-11-15:
Looks like someone at Creality re-compiled this version.  I downloaded it again (it says from 3 days ago) and it works now on my system that still has openssl 3.

JaneSmith, 2024-11-15:
> Looks like someone at Creality re-compiled this version. I downloaded it again (it says from 3 days ago) and it works now on my system that still has openssl 3.

Huh. You're right. That's very strange. It has the exact same filename, but the checksum is different — it was just silently swapped out? I can confirm that this new version of the same version works in my Fedora Distrobox now.

Goodolandy, 2024-11-18:
Still crashing for me on launch using Linux Mint 21.3.  Now my printer, Sermoon V1 Pro, still isn't supported in v5 but would be nice to upgrade from 4.3.6.6257.

AtomicRobotMan0101, 2024-11-18:
@Goodolandy I had the same problem when using the Nvidia drivers.

Use the "Driver Manager" tool and change to Nouveau. 

One can see the actual crash and the error it creates if you use the Terminal to start the app... i.e `./Creality_Print-v5.1.6.10470-x86_64-Release.AppImage` .... the terminal outputs a bunch of spooge showing it exploding.

I tried many things, including an extensive debugging to solve it under nvidia, but proved unable to. Changing the driver worked.

Goodolandy, 2024-11-18:
@AtomicRobotMan0101 Thank you.  I never thought the problem would be a graphics driver.  I saw that I was still on nvidia-driver-535.  Updated to 550 and now Creality Print v5 is loading.  I hope they start adding more printers soon.

wbvczar, 2024-11-18:
I found that the current version of Creality Print for Linux doesn't like something about Wayland with Nvidia drivers.  I could run it in Wayland with the default video drivers, but once it was sliced, I couldn't rotate or move around the model as it was really lagging.  I installed the Nvidia drivers and then Creality Print would crash on startup.  I changed back to X11 and Creality loads fine with the Nvidia drivers.

I am running most recent KDE Neon User Edition with Nvidia GTX 1080Ti.

kanocz, 2024-11-22:
tested on kde neon (based on ubuntu 24.04) and new uploaded version works

basilverthorn, 2024-11-26:
Same issue here, but I am on ARCH. Tried almost all of the above no go. I went back 4.3, it runs flawlessly for now.

quevon24, 2024-11-27:
> Running on 24.04, x11, and nvidia gpu with the default openssl 3.0.13 and the following flag:
>
> ```
> __NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=nvidia [...]

This also works in linux mint 22 cinnamon

basilverthorn, 2024-11-27:
Thank you, I saw this on another post. Does not seem to fix anything here unfortnately.  For the life me, I can't figure this out. 4.3 works great. But I am on ARCH not ubuntu as well.

Output:

Exec=env __NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=nvidia ./Creality_Print-v5.1.6.10470-x86_64-Release_73540cdc021c8c1465f106722cd6f499.appimage
QFont::fromString: Invalid description 'Fira Sans,10,-1,5,400,0,0,0,0,0,0,0,0,0,0,1'
QFont::fromString: Invalid description 'FiraCode Nerd Font Mono,10,-1,5,400,0,0,0,0,0,0,0,0,0,0,1'
QFont::fromString: Invalid description 'Fira Sans,10,-1,5,400,0,0,0,0,0,0,0,0,0,0,1'
QFont::fromString: Invalid description 'Fira Sans,10,-1,5,400,0,0,0,0,0,0,0,0,0,0,1'
Initializing StaticPrintConfigsCXBinManager open file size [377984]
parse binary stl ...face 7558

parse binary stl success...

Computing bounding box... Done.
x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
Computing bounding box... Done.
x = -57.2188 .. 57.2205, y = -49.9992 .. 50.0008, z = -0.0766628 .. 228.967
Computing bounding box... Done.
x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
Computing bounding box... Done.
x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
X connection to :1 broken (explicit kill or server shutdown).

basilverthorn, 2024-11-27:
Seems top be an issue on Wayland, if I log into an ancient X11 session it works. Which is doable, but not preferred, this is the only app that I have that won't run in Wayland I guess. Orca works fine as of 2.2. I live in Wayland and Hyprland. Currenttly I just 4.3, as it works for now.

metracy, 2024-11-28:
> This also works in linux mint 22 cinnamon

Worked well in Xubuntu 22.04 and NVidia 550 drivers

basilverthorn, 2024-11-28:
Anyone have it working in Wayland? Like 4.3 does?

JaneSmith, 2024-11-28:
Works just fine for me on Wayland in a Fedora Distrobox. I see a lot of people talking about Nvidia, so that's probably the issue — it's well-known that Nvidia doesn't have great Wayland support. I'm using an AMD card.

basilverthorn, 2024-11-28:
Ahh that could be, 4.3 works great. My guess is that it's missing a library
or something. 4.3 works in wayland fine.
Thanks for the response. Bambu works fine in wayland and Orca. I can stay
in those two, there just some times I want creality print.

Goodolandy, 2024-11-28:
I would be up for using the Orca slicer, but it doesn't have the Sermoon v1 Pro printer profile. I tried the standard v1 profile, and my printer went crazy, so I had to stop the printing immediately before any damage was done.  Hopefully, a proper v1 Pro profile will be added to Creality Print v5 or Orca.

butfar, 2024-12-13:
Mint 22 ca

> @Goodolandy I had the same problem when using the Nvidia drivers.
> 
> Use the "Driver Manager" tool and change to Nouveau.
> 
> One can see the actual crash and the error it creates if you use the Terminal to start the app... i.e `./Creality_Print-v5.1.6.10470-x86_64-Release.AppImage` .... the terminal outputs a bunch of spooge showing it exploding.
> 
> I tried many things, including an extensive debugging to solve it under nvidia, but proved unable to. Changing the driver worked.
> 
> (edit - Using Mint 22, fully updated. No weird setups, all very generic)

Changing the nvidia driver helped me

BloodyIron, 2024-12-15:
DOWNGRADING YOUR OPENSSL VERSION FROM 3.x to 1.1 IS REALLY A BAD SECURITY IDEA. GUYS, DON'T DO THAT, YOU'RE MAKING YOUR SYSTEM SUBSTANTIALLY LESS SECURE.

BB31420, 2024-12-18:
> **DOWNGRADING YOUR OPENSSL VERSION FROM 3.x to 1.1 IS REALLY A BAD SECURITY IDEA. GUYS, DON'T DO THAT, YOU'RE MAKING YOUR SYSTEM SUBSTANTIALLY LESS SECURE.**

Rest assured that the Ubuntu security team will continue to maintain important security fixes in OpenSSL 1.1.1 for as long as [the Ubuntu release is supported](https://ubuntu.com/about/release-cycle), meaning 2025 for Ubuntu 20.04 LTS with standard support. For [Ubuntu Pro subscribers](https://ubuntu.com/pro), Canonical offers Expanded Security Maintenance (ESM). This means getting at least 10 years of security maintenance for all software [bundled in the release](https://launchpad.net/ubuntu), including OpenSSL 1.1.1. This translates to being security maintained and supported until at least 2028 for Ubuntu 18.04 LTS and at least 2030 for Ubuntu 20.04 LTS.

Make sure to activate your free one per household **UBUNTU PRO MEMBERSHIP** which isn't necessary but might not know it's there for free.

wbvczar, 2024-12-18:
> Rest assured that the Ubuntu security team will continue to maintain important security fixes in OpenSSL 1.1.1 for as long as [the Ubuntu release is [...]

This is good information.  Also, with the updated appimage (make sure to download again to get the latest) it was recompiled and no longer needs openssl 1.1.  It works great with the latest openssl with no downgrade necessary.

butfar, 2024-12-18:
The problem with the interaction of the NVideo Geforce 1650 video card in the latest release 3DCreality_Print-v5.1.7.10514-x86_64-Release.AppImage for MintLinux has not been resolved. The problem is that the program opens and closes immediately, and if you roll back the drivers on Linux, the program works.

JimB2020, 2024-12-22:
I was so happy to get Creality Print 5.1.6 to work on my Linux Mint 21.3 Cinnamon computer using the command line entry posted by BB31420 (thank you!), but unfortunately it does not work on 5.1.7.  The splash screen comes up and hangs for a little while and then disappears.  Here is the output I get in the terminal:

>  _NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=nvidia ./Creality_Print-v5.1.7.10514-x86_64-Release.AppImage
> Initializing StaticPrintConfigsCXBinManager open file size [377984]
> parse binary stl ...face 7558
> 
> parse binary stl success...
> 
> Computing bounding box... Done.
>   x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
> Computing bounding box... Done.
>   x = -57.2188 .. 57.2205, y = -49.9992 .. 50.0008, z = -0.0766628 .. 228.967
> Computing bounding box... Done.
>   x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
> Computing bounding box... Done.
>   x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
> XIO:  fatal IO error 2 (No such file or directory) on X server ":0"
>       after 747 requests (747 known processed) with 0 events remaining.
> Segmentation fault (core dumped)

I went back and tried it with Creality Print 5.1.6 just to make sure it still works, but it too crashes in much the same way now.  It was working just fine for me last Tuesday, so the only thing I can think of is that a recent kernel update or an NVIDIA driver update broke the solution that BB31420 figured out.  For what its worth, here is my current kernel and NVIDIA info:

> 
> Linux kernel:  5.15.0-69-generic
> NVIDIA Corporation GK208B [GeForce GT 730]
> NVIDIA Driver Version:  470.256.02

basilverthorn, 2024-12-22:
Try replacing nvidia with amd in the command line.  It’s broke here too now, can’t either version to work. Only 4.3 now.  Sent from my iPhone

BB31420, 2024-12-25:
> I was so happy to get Creality Print 5.1.6 to work on my Linux Mint 21.3 Cinnamon computer using the command line entry posted by @BB31420 (thank you!), but unfortunately it does not work on 5.1.7. [...]

Jiim - happy holidays - @AtomicRobotMan0101 suggested using `strace -o error.txt ./Creality_Print-v5.1.7.10514-x86_64-Release.AppImage` to capture error output in a text file to read.

Any chance you had tried to set a temp environment variable for downgrading openssl  and it has reset upon a new session? It sounds like you only needed the nvidia flag before tho but just throwing this out there. Initially downgrading openssl allowed me to move forward in the debugging process to find the other flag.

Good lucks, let me know more about this error if you can capture verbose output

Cheers

JimB2020, 2024-12-26:
@BB31420 - happy holidays!

> - @AtomicRobotMan0101 suggested using strace -o error.txt ./Creality_Print-v5.1.7.10514-x86_64-Release.AppImage to capture error output in a text file to read.

Holy smokes!  The strace error.txt file is huge - 3.2MB!  I tried looking through some of it, but in addition to it being such a massive amount of data, I'm not sure what I'm looking for...

I tried most of the items on your testing commands list skipping things that were not in my current repositories (such as nvidia-opengl-dev), and a few things that I was afraid might break something.  I found that have OpenSSL 3.0.2, but I also found that I have libssl1.1.1f-1ubuntu2 already installed.  I'm not sure how to 

> set a temp environment variable for downgrading openssl

but by the time i got down to trying line 1991 on your testing commands list (the original workaround of the nvidia flag) - it worked!  Being suspicious of my inexplicable positive results, I rebooted my computer and tried it again.  No luck.  Same errors.

> perhaps it's not finding the primary monitor.

Primary monitor is correct - checking it both from the command line and the NVIDIA X SERVER SETTINGS on the Administration menu.

Thanks for your help

AtomicRobotMan0101, 2024-12-26:
@JimB2020 you are not wrong about the output, its a bit much.

The fix turns out to be easier than what I suggested.... took a while to work out!

- edit the `/etc/environment` file
- `sudo nano /etc/environment`
- add this: `WEBKIT_DISABLE_DMABUF_RENDERER=1`
- save the file
- log out of your session (or reboot)

The CrealityPrint will now work.

basilverthorn, 2024-12-26:
This was a no go for me ;-(. Darn it! Just starts and immediately exists.
What is the variable to alter to make it think you're not on the latest SSL?

JimB2020, 2024-12-26:
> This was a no go for me ;-(. Darn it! Just starts and immediately exists.
> What is the variable to alter to make it think you're not on the latest SSL?

@AtomicRobotMan0101's  the solution of editing the `/etc/environment` file helped me find a solution that seems get the nvidia flag to work every time - just put the environment command at the beginning.

`env __NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=nvidia ./Creality_Print-v5.1.7.10514-x86_64-Release.AppImage`

I've rebooted twice just to be sure this always works, and yes it does!

v3rtx, 2024-12-28:
> Worked well in Xubuntu 22.04 and NVidia 550 drivers

Thanks mate! Works like a charm!

basilverthorn, 2024-12-28:
Awesome fix Ubuntu.  Still no go on arch. Sent from my iPhone

BB31420, 2024-12-30:
You're running Wayland on Arch with an AMD gpu and is it external or integrated or part of the CPU? I will try to fix this for you, let me know more about your build
happy new year

basilverthorn, 2024-12-30:
I’m running Wayland, on Arch, with Nvidia GPU, core i9 intel cpu. Some have reported changing the nvidia to amd worked for them.  The are a few of us on the creality discord channel all experiencing the same thing.  It’s hard to tell if it’s a Wayland thing or Nvidia thing. Sent from my iPhone

baruchsheffi, 2025-01-01:
I changed the flag to amd even though i have nVidia 1650 and it started working... 
#!/bin/bash
__NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=amd ./Creality_Print-v5.1.7.10514-x86_64-Release.AppImage

at least it works now

BB31420, 2025-01-02:
```

sudo pacman -S nvidia-utils lib32-nvidia-utils
```
force x11 
```
GDK_BACKEND=x11 __NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=nvidia ./Creality_Print-v5.1.7.10514-x86_64-Release.AppImage
```
force x11 with AMD's GLX library
```
GDK_BACKEND=x11 __NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=AMD ./Creality_Print-v5.1.7.10514-x86_64-Release.AppImage
```

The user's where changing to AMD's GLX library worked may have x11 setups. If none of this works, I have a copy of arch to setup and test but might be a few days.

basilverthorn, 2025-01-03:
Differnt error now, like no error. It does not even show the screen now: Executes per say, no splash screen, then back to prompt.

╭─brett@I9 in ~/Applications as 🧙 took 0s
╰─λ GDK_BACKEND=x11 __NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=nvidia ./Creality-Print_73540cdc021c8c1465f106722cd6f499.AppImage
Initializing StaticPrintConfigsCXBinManager open file size [377984]
parse binary stl ...face 7558

parse binary stl success...

Computing bounding box... Done.
x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
Computing bounding box... Done.
x = -57.2188 .. 57.2205, y = -49.9992 .. 50.0008, z = -0.0766628 .. 228.967
Computing bounding box... Done.
x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
Computing bounding box... Done.
x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387

╭─brett@I9 in ~/Applications as 🧙 took 1s
╰─λ GDK_BACKEND=x11 __NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=amd ./Creality-Print_73540cdc021c8c1465f106722cd6f499.AppImage
Initializing StaticPrintConfigsCXBinManager open file size [377984]
parse binary stl ...face 7558

parse binary stl success...

Computing bounding box... Done.
x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
Computing bounding box... Done.
x = -57.2188 .. 57.2205, y = -49.9992 .. 50.0008, z = -0.0766628 .. 228.967
Computing bounding box... Done.
x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
Computing bounding box... Done.
x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387

Crono141, 2025-01-27:
I just encountered this same error, running Fedora 41 (Well, Nobara 41), and also have a 1650 video card. Running on wayland with a segmentation fault using this command:

`GDK_BACKEND=x11 __NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=nvidia ./creality_print.appimage
Initializing StaticPrintConfigsCXBinManager open file size [377984]
parse binary stl ...face 7558

parse binary stl success...

Computing bounding box... Done.
  x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
Computing bounding box... Done.
  x = -57.2188 .. 57.2205, y = -49.9992 .. 50.0008, z = -0.0766628 .. 228.967
Computing bounding box... Done.
  x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
Computing bounding box... Done.
  x = -3.4333 .. 3.4334, y = -3.0001 .. 3.0002, z = -0.0046 .. 13.7387
X connection to :0 broken (explicit kill or server shutdown).
Segmentation fault (core dumped)
`

basilverthorn, 2025-01-27:
Same I tried Nobara as well by recommendation.  Love the distro, but same error here too.  Sent from my iPhone

closed 2025-05-05 by github-actions[bot]