CrealityOfficial/CrealityPrint #235: Linux aarch64

TheNightmanCodeth, 2024-11-11:
Latest (5.1.6) release is missing an aarch64 (arm64) linux build. Any chance someone at Creality could get a build up for us lowly arm folks? 

I would gladly do so myself but the sources are not yet public :broken_heart:

aallbrig, 2024-11-11:
I can't find any linux builds for any of the version 5 of creality pint. I'm still using creality print version 4 but would love to be able to remove brims (a feature in version 5).

Just adding my voice as a party very interested in being able to use the latest software on linux.

dwaite, 2024-11-18:
There are apple silicon builds as of 5.1.6 under the releases section here.

TheNightmanCodeth, 2024-11-23:
@dwaite yeah, they don't work on linux though (;

closed 2025-04-27 by github-actions[bot]